jrowberg/i2cdevlib #332: MPU6050_DMP6 example gives erratic values (MPU6050_RAW works great)

pepelepoisson, 2017-10-05:
Hi,
First of all **MANY THANKS** for this great work. I am experimenting with MPU6050 library (september 23rd version, downloaded yesterday) using an Arduino Mega 2560 and a GY-521 board. MPU6050_RAW works fine. I2C scanner indicates 0x68 address as expected. I have problems with MPU6050_DMP6 example. Code starts OK but gives inconsistent numbers (random values in spite of the fact that the accel is stable on the table), shows FIFO overflow on a few occasions and freezes after about one minute. Details of wiring and serial output copy are shown below for reference. Do you know what is causing this issue? Any test I could make to help narrowing the problem????? Any previous post on this topic maybe???? Many thanks for your help and best regards.

Pascal Prado.

Wiring:
=====
Mega 2560 3.3V > GY-521 VCC
Mega 2560 GND > GY-521 GND
Mega 2560 GND > GY-521 AD0
Mega 2560 D21 > GY-521 SCL
Mega 2560 D20 > GY-521 SDA
Mega 2560 D2 > GY-521 INT

Serial print output:
==============
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful

Send any character to begin DMP programming and demo: 
Initializing DMP...
Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
**ypr    47.00    5.65    -1.16
ypr    137.34    7.39    31.24
ypr    36.56    46.02    41.07
ypr    180.00    -0.33    -0.34
ypr    72.22    44.80    37.85**
ypr    -104.91    65.12    10.01
ypr    -179.94    -15.27    0.63
ypr    -139.03    -14.73    12.46
ypr    2.04    -4.61    47.25
ypr    -58.69    69.70    -17.63
ypr    -50.45    -84.94    4.99
ypr    180.00    0.16    -0.56
ypr    27.51    -59.79    13.81
ypr    -16.97    -54.16    9.76
ypr    167.23    3.89    59.01
ypr    88.06    29.17    60.79
ypr    -179.99    0.04    13.09
ypr    64.17    12.19    -2.73
ypr    -0.23    6.10    32.27
ypr    -180.00    -0.13    -4.59
ypr    7.52    0.76    -25.99
ypr    48.46    6.35    -0.66
ypr    89.90    -24.84    -45.26
ypr    -26.24    -44.63    32.17
ypr    180.00    -0.28    -0.29
ypr    17.63    -9.72    -80.25
ypr    0.43    66.66    -4.15
ypr    -179.93    -14.81    0.55
ypr    41.64    -8.68    24.52
ypr    2.42    -5.29    48.67
ypr    38.86    -50.68    24.01
ypr    24.31    -11.19    -73.76
ypr    180.00    0.15    -0.53
ypr    -81.73    -9.25    57.27
ypr    179.18    6.69    49.13
ypr    166.94    4.01    60.27
ypr    123.13    13.91    71.80
ypr    -179.99    0.04    14.45
ypr    -37.30    49.89    25.36
ypr    51.53    -18.07    18.02
ypr    -180.00    -0.13    -4.87
ypr    -4.26    5.81    -64.68
ypr    49.96    6.96    -0.21
ypr    89.64    -12.41    -20.89
ypr    -53.77    49.36    -36.43
ypr    180.00    -0.25    -0.26
ypr    -102.18    -77.17    12.57
ypr    15.12    25.83    -63.08
ypr    -179.93    -14.83    0.51
ypr    -86.59    -49.31    10.57
ypr    2.77    -5.85    50.09
ypr    -54.35    76.02    -7.92
ypr    -39.79    65.55    -3.06
ypr    180.00    0.15    -0.53
ypr    -19.04    -16.13    9.47
ypr    -172.29    -11.49    49.48
ypr    166.90    4.03    60.28
ypr    107.03    27.60    11.07
ypr    -179.99    0.04    15.46
ypr    -22.50    -34.54    53.16
ypr    48.90    9.10    -7.47
ypr    -180.00    -0.13    -4.31
ypr    -29.86    33.91    -13.01
ypr    51.46    7.46    0.17
ypr    89.69    5.06    9.25
ypr    -15.46    28.13    8.28
ypr    180.00    -0.23    -0.24
ypr    11.26    17.36    -69.43
ypr    6.02    -57.54    10.12
ypr    -179.92    -14.78    0.63
ypr    -47.87    -40.36    -11.63
ypr    3.08    -6.29    51.50
ypr    47.91    -39.68    20.17
ypr    10.99    -21.63    66.92
ypr    180.00    0.15    -0.53
ypr    25.73    45.25    -44.72
ypr    -20.78    -56.12    12.26
ypr    166.87    4.05    60.30
ypr    147.84    9.92    80.07
ypr    -179.99    0.04    16.18
ypr    -33.68    35.85    42.69
ypr    -51.33    41.09    23.80
ypr    -180.00    -0.13    -4.20
ypr    -175.27    1.07    -75.59
ypr    52.95    7.87    0.51
ypr    89.57    18.10    32.56
ypr    43.66    -72.33    -17.64
ypr    180.00    -0.22    -0.22
ypr    -6.08    6.34    73.33
ypr    79.06    9.23    -11.46
ypr    -179.93    -14.77    0.53
ypr    27.15    -2.64    50.67
ypr    3.36    -6.66    52.89
ypr    -62.26    64.26    -23.97
ypr    39.64    40.89    -22.80
ypr    180.00    0.15    -0.53
ypr    38.38    -79.08    -0.59
ypr    166.17    26.43    49.33
ypr    166.94    4.04    60.34
**FIFO overflow!
FIFO overflow!**
ypr    56.27    8.53    1.09
ypr    89.62    -30.54    -59.45
ypr    -44.28    42.09    -16.41
ypr    180.00    -0.20    -0.20
ypr    41.72    -39.32    -7.54

pepelepoisson, 2017-10-06:
Quick update. 
(1) The erratic numbers and FIFO overflow problems disappeared when I updated my I2Cdev library (I had a version from 2011 now I have the latest available today). Numbers now look OK.
(2) Freeze problem remains. To be investigated further.

dldelaney, 2019-08-02:
I am having this same problem - when adding a delay as low as 6 ms at the very bottom of the code, it starts spitting out erratic values.  
I've also had the freezing issue where the Arduino seems to just stop looping (no commands are run (print statements, LED turn on/off, etc)), sometimes after just a few seconds (one time it crashed less than a second after starting), sometimes not at all.  I can't seem to get repeatable results for crashing the Arduino.

(All this was using the MPU6050_DM6 example code on an Arduino Uno using a fresh download of the I2C and MPU6050 libraries on 8/2/19)
VCC->5V
GND->GND
SCL->A5
SDA->A4
INT->2