jrowberg/i2cdevlib #338: Reliability is related to value set on BUFFER_LENGTH for ESP32

rummyr, 2017-11-20:
This is a more of a let-you-know.
If I use MPU6050-DMP6 with an ESP32 I get some "interesting" reliability issues depending on the value I use for #define BUFFER_LENGTH  @ the top of I2Cdev.h
@16 the code runs and spits out quat and aworld reliably for >10 minutes
@32 it runs OK, unless I "tap" the MPU
@I2C_BUFFER_LENGTH (128 I think) it dies after just a couple of readings (if it even starts)

Although this is an "issue" I'm not asking you to fix it, just letting you know.

arvidtp, 2017-11-21:
I'm working on this too with the sparkfun ESP32 Thing board. Haven't tested enough yet to corroborate about reliability, but I couldn't connect at all without reducing i2c frequency from 400kHz to 100kHz. Have you done that?

But thanks - I'll keep this in mind as I test. My application needs to be reliable :)

arvidtp, 2017-12-03:
The combo of 100kHz i2c freq and 128 BUFFER_LENGTH seems to be best for me. MPU has been running for at least 15 min, including "tapping" on it. Not sure its 100kHz - but just leaving the i2c freq at its default instead of setting to 400kHz.

carbonadam, 2018-01-13:
im not having any luck at all with this lib and the esp32, no mpu6050 for me..anyone who manages to get it to work with the arduino IDE let us know how:)

arvidtp, 2018-01-14:
To get it to be stable I had to put everything all the i2c communication it its own task on core 1.

Here's my code: https://gist.github.com/arvidtp/d0ab689a5c2821a45c31103d7a3e256a

it's a little messy, but should give you the idea.