jrowberg/i2cdevlib #341: Build error with Arduino ESP8266 staging version

me21, 2017-12-01:
```
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: error: no matching function for call to 'min(uint8_t&, int)'
for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {
^
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: note: candidates are:
In file included from c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,
from C:\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/Arduino.h:240,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.h:80,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:46:
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Comp
are)
min(initializer_list<_Tp> __l, _Compare __comp)
^

c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {
^
In file included from c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,
from C:\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/Arduino.h:240,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.h:80,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:46:
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)
min(initializer_list<_Tp> __l)
^
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {
^
In file included from c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,
from C:\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/Arduino.h:240,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.h:80,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:46:
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&,
 _Compare)
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {
^
In file included from c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,
from C:\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/Arduino.h:240,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.h:80,
from .piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:46:
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)
min(const _Tp& __a, const _Tp& __b)
^
c:\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:
.piolibdeps\I2Cdevlib-Core_ID11\I2Cdev.cpp:276:75: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {
^
*** [.pioenvs\esp01_1m\lib\I2Cdevlib-Core_ID11\I2Cdev.o] Error 1
```

muratozkan, 2018-01-10:
I should note that it is not a staging version anymore and occurs when compiling against Espressif 8266 platform `1.6.0` (it was compiling fine on `1.5.0`).

jrowberg, 2018-03-04:
Are `_min()` and `_max()` available on common non-ESP platforms? I am only concerned with making sure this doesn't break other implementations.

me21, 2018-03-04:
You're right, it's not, at least for STM32. My bad. As there are conflicting types in template arguments deducing, perhaps explicit type casts will help instead?

halfcadence, 2018-03-09:
I was able to compile and run on esp8266 by forking and replacing the min() function:
https://github.com/halfcadence/i2cdevlib/commit/94677116c62aff6a2ce08d96522d3a8e89d12b4b

Obviously there's some problems with this solution but if you just need a hack to get things to work...

tiger-greyhawk, 2018-03-09:
<b>halfcadence</b>, i dont know about problems, but it worked for me on WeMos (based esp8266) with MPU5060. Perhaps i will to have wrongs in the future, but... 
Very many thanks at this time.

Ravenheart, 2018-03-24:
DOIT DEVKIT V3 here with latest i2cdev/mpu6050/esp8266 arduino core and it fails with the following:

[gist](https://gist.github.com/Ravenheart/f9add6fbf5e90518c4d31e6c30245622)

ingeltro, 2018-03-28:
Hi, I have the same problem as "Ravenheart" ....

Zarfab, 2018-07-10:
Hi, 
I know this is far to be the cleanest solution but I managed to solve the problem by adding 
`#ifdef ESP8266
	#define min _min
	#define max _max
#endif`
and 
`#define BUFFER_LENGTH 64`
in I2Cdev.h, just before I2Cdev class declaration

I also modified line 50 in MPU6050_6Axis_MotionApps20.h and MPU6050_9Axis_MotionApps41.h to replace it with
`#if !defined(__PGMSPACE_H_) && !defined(PGMSPACE_INCLUDE)`
I now compiles and runs on both ESP8266 and ESP32

ingeltro, 2018-07-10:
Zarfab Thanks, coincidentally I had also arrived at that solution after much trial and error.

commit c20f68ec (2018-09-12): Merge pull request #342 from me21/master

Fix #341.
commit ceba54af (2018-09-12): Fix #341.
closed 2018-09-12 by jrowberg