jrowberg/i2cdevlib #346: Problems with MPU6050 and ESP32

jollepe, 2017-12-19:
I tried to connect the MPU6050 with my WeMos ESP32 Dev-Board via I2C.
So I copied the ESP32 related files from this repository to my Arduino library folder and started the example MPU6050_DMP6. During compiling I got these errors.
What should I do to get things running?

```
In file included from /tmp/arduino_modified_sketch_453924/MPU6050_DMP6.ino:48:0:
/home/julian/Arduino/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:66:24: error: conflicting declaration 'typedef int8_t prog_int8_t'
         typedef int8_t prog_int8_t;
                        ^
In file included from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/WString.h:29:0,
                 from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:150,
                 from sketch/MPU6050_DMP6.ino.cpp:1:
/home/julian/Arduino/hardware/espressif/esp32/cores/esp32/pgmspace.h:25:14: note: previous declaration as 'typedef char prog_int8_t'
 typedef char prog_int8_t;
              ^
In file included from /tmp/arduino_modified_sketch_453924/MPU6050_DMP6.ino:48:0:
/home/julian/Arduino/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:70:25: error: conflicting declaration 'typedef int32_t prog_int32_t'
         typedef int32_t prog_int32_t;
                         ^
In file included from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/WString.h:29:0,
                 from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:150,
                 from sketch/MPU6050_DMP6.ino.cpp:1:
/home/julian/Arduino/hardware/espressif/esp32/cores/esp32/pgmspace.h:29:14: note: previous declaration as 'typedef long int prog_int32_t'
 typedef long prog_int32_t;
              ^
In file included from /tmp/arduino_modified_sketch_453924/MPU6050_DMP6.ino:48:0:
/home/julian/Arduino/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:71:26: error: conflicting declaration 'typedef uint32_t prog_uint32_t'
         typedef uint32_t prog_uint32_t;
                          ^
In file included from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/WString.h:29:0,
                 from /home/julian/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:150,
                 from sketch/MPU6050_DMP6.ino.cpp:1:
/home/julian/Arduino/hardware/espressif/esp32/cores/esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'
 typedef unsigned long prog_uint32_t;
                       ^
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

herm, 2018-01-25:
Removing these lines from MPU6050_6Axis_MotionApps20.h should solve the problem:
```
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
```
Or even better wrap them in an approriate #ifdef and submit a pull request.

ferasterkawi, 2019-01-25:
I tried to program ESP32 with Arduino IDE but I faced this problems:
What should I do?

Arduino: 1.8.1 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, QIO, 80MHz, 4MB (32Mb), 256000, None"

C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:142:57: error: invalid conversion from 'int' to 'i2c_ack_type_t' [-fpermissive]

   ESP_ERROR_CHECK(i2c_master_read(cmd, data, length-1, 0));

                                                         ^

C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:49:51: note: in definition of macro 'ESP_ERROR_CHECK'

 #define ESP_ERROR_CHECK(x)   do { esp_err_t rc = (x); if (rc != ESP_OK) { ESP_LOGE("err", "esp_err_t = %d", rc); /*assert(0 && #x);*/} } while(0);

                                                   ^

In file included from C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.h:37:0,

                 from C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:40:

C:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver/driver/i2c.h:325:11: note:   initializing argument 4 of 'esp_err_t i2c_master_read(i2c_cmd_handle_t, uint8_t*, size_t, i2c_ack_type_t)'

 esp_err_t i2c_master_read(i2c_cmd_handle_t cmd_handle, uint8_t* data, size_t data_len, i2c_ack_type_t ack);

           ^

C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:144:60: error: invalid conversion from 'int' to 'i2c_ack_type_t' [-fpermissive]

  ESP_ERROR_CHECK(i2c_master_read_byte(cmd, data+length-1, 1));

                                                            ^

C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:49:51: note: in definition of macro 'ESP_ERROR_CHECK'

 #define ESP_ERROR_CHECK(x)   do { esp_err_t rc = (x); if (rc != ESP_OK) { ESP_LOGE("err", "esp_err_t = %d", rc); /*assert(0 && #x);*/} } while(0);

                                                   ^

In file included from C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.h:37:0,

                 from C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\I2Cdev\I2Cdev.cpp:40:

C:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver/driver/i2c.h:306:11: note:   initializing argument 3 of 'esp_err_t i2c_master_read_byte(i2c_cmd_handle_t, uint8_t*, i2c_ack_type_t)'

 esp_err_t i2c_master_read_byte(i2c_cmd_handle_t cmd_handle, uint8_t* data, i2c_ack_type_t ack);

           ^

exit status 1
Error compiling for board ESP32 Dev Module.
Exception in thread "Thread-111" java.lang.NullPointerException
	at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:114)
	at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:106)
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:665)
	at processing.app.Base.onBoardOrPortChange(Base.java:1295)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2220)
	at java.lang.Thread.run(Thread.java:745)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

A4lfr32, 2020-04-24:
> Removing these lines from MPU6050_6Axis_MotionApps20.h should solve the problem:
>
> ```
>         typedef int8_t prog_int8_t;
>         typedef uint8_t prog_uint8_t;
>         typedef int16_t [...]

work for me.