jrowberg/i2cdevlib #353: ADS1115 low data rate

MiracelVip, 2018-01-26:
Hello,

I have a problem with the data rate of the ADS1115. It Should reach up to 860 sps but for me the maximum is ~200 sps. Where may be the limit of the data rate?

I'm doing a simple test like:

```
adc0.setMode(ADS1115_MODE_SINGLESHOT);
adc0.setRate(ADS1115_RATE_860);

 do {
    // The below method sets the mux and gets a reading.
    adc0.setMultiplexer(ADS1115_MUX_P0_NG);
    adc0.triggerConversion();
    pollAlertReadyPin();
    float val1 = adc0.getMilliVolts(false);
    
    adc0.setMultiplexer(ADS1115_MUX_P1_NG);
    adc0.triggerConversion();
    pollAlertReadyPin();
    float val2 = adc0.getMilliVolts(false);
    
    adc0.setMultiplexer(ADS1115_MUX_P2_NG);
    adc0.triggerConversion();
    pollAlertReadyPin();
    float val3 = adc0.getMilliVolts(false);
    
    adc0.setMultiplexer(ADS1115_MUX_P3_NG);
    adc0.triggerConversion();
    pollAlertReadyPin();
    float val4 = adc0.getMilliVolts(false);
    
    count++;
  }while((millis() - startTime) < 1000);
  Serial.print("Reached rate = ");Serial.println(count*4);
```

and get:

>  Testing device connections...
> ADS1115 connection successful
> 0
> Reached rate = 216

I'm using a Arduino Pro Mini + ADS1115

Thank you in advance

MiracelVip, 2018-01-26:
Wire.setClock(400000); increased the rate to 472 sps

lacklustrlabs, 2018-01-26:
I would test with a little bit faster MCU, even 472 sps can be taxing for a 8Mhz pro mini. 
At least just for isolating the bottleneck. ADS1115 is 5V tolerant, so any old UNO should work.

alijani1, 2020-05-23:
for me, whats odd is when I only read one port instead of all 4 it you will get <1ms  per read but if i read all 4 ports sequentially, it takes total time of >11ms  instead of staying what I would expect to be <4ms.  I dont understand why reading multiple ports is so much slower.