jrowberg/i2cdevlib #368: MPU6050+ESP32+Arduino

vinicius121, 2018-04-06:
Hi,

I have a Wemos Lolin32 board and I'm trying to compile the MPU6050_DMP6 example on the Arduino IDE but with no success.
When compiling I have the following erros:
`
In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:56:0: warning: "PSTR" redefined

         #define PSTR(str) (str)

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:36:0: note: this is the location of the previous definition

 #define PSTR(s)       (s)

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:57:0: warning: "F" redefined

         #define F(x) x

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150:0,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:38:0: note: this is the location of the previous definition

 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:69:0: warning: "strcpy_P" redefined

         #define strcpy_P(dest, src) strcpy((dest), (src))

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:72:0: note: this is the location of the previous definition

 #define strcpy_P      strcpy

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:71:0: warning: "strcmp_P" redefined

         #define strcmp_P(a, b) strcmp((a), (b))

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:76:0: note: this is the location of the previous definition

 #define strcmp_P      strcmp

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:74:0: warning: "pgm_read_word" redefined

         #define pgm_read_word(addr) (*(const unsigned short *)(addr))

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:40:0: note: this is the location of the previous definition

 #define pgm_read_word(addr) ({ \

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:75:0: warning: "pgm_read_dword" redefined

         #define pgm_read_dword(addr) (*(const unsigned long *)(addr))

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:44:0: note: this is the location of the previous definition

 #define pgm_read_dword(addr) ({ \

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:76:0: warning: "pgm_read_float" redefined

         #define pgm_read_float(addr) (*(const float *)(addr))

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:48:0: note: this is the location of the previous definition

 #define pgm_read_float(addr) ({ \

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:62:24: error: conflicting declaration 'typedef int8_t prog_int8_t'

         typedef int8_t prog_int8_t;

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:25:14: note: previous declaration as 'typedef char prog_int8_t'

 typedef char prog_int8_t;

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:66:25: error: conflicting declaration 'typedef int32_t prog_int32_t'

         typedef int32_t prog_int32_t;

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:29:14: note: previous declaration as 'typedef long int prog_int32_t'

 typedef long prog_int32_t;

In file included from C:\Users\vinic\Documents\Arduino\libraries\MPU6050\Examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\vinic\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:67:26: error: conflicting declaration 'typedef uint32_t prog_uint32_t'

         typedef uint32_t prog_uint32_t;

In file included from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:150,

                 from sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\vinic\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'

 typedef unsigned long prog_uint32_t;

exit status 1
Error compiling for board WEMOS LOLIN32.`

Anyone successfully got this working? If so can you share your project?

Thanks,
Vinicius

vinicius121, 2018-04-06:
Hi,

I got it working.
What I did was edit two .h files as described here: https://github.com/jrowberg/i2cdevlib/pull/367
Then commented the above lines inside the #ifdef __AVR___
`typedef int8_t prog_int8_t;
 typedef uint8_t prog_uint8_t;
 typedef int16_t prog_int16_t;
 typedef uint16_t prog_uint16_t;
 typedef int32_t prog_int32_t;
 typedef uint32_t prog_uint32_t;`

And maybe for better results see #338 .

Thanks,
Vinicius

cruizg, 2019-04-15:
@vinicius121 can you shared your connection please.

vinicius121, 2019-04-16:
@cruizg what do you mean?

cruizg, 2019-04-16:
@vinicius121  How did you make the connections in your Wemos Lolin32? thanks a lot.

intensite, 2019-11-28:
What is the status of the ESP32 port as of end of 2019?

I am having a hard time compiling the I2Cdev and MPU6050.  I followed the recommendations stated above without much success.  I fix one compile error but 10 more appears :-(

Everything was working fine on an arduino nano until I ran out of memory, that is why I am hoping to make it work with an Adafruit Huzzah32 Feather.

Is it possible to use the library with this board?

Regards.

Stephen..

jrowberg, 2021-09-28:
This should compile correctly on ESP32 boards in the Arduino IDE as of the latest commit, including the Wemos Lolin32. I don't have a board to test runtime functionality, but I expect it will work.

closed 2021-09-28 by jrowberg

mark-hahn, 2021-11-24:
I hope this isn't a duplicate post.  I thought I replied before.

I also have not been able to get the 6050 example to compile for esp32 using the latest clone of this repo.  I'm using platformio which should be the same as the arduino ide.  You say it should compile correctly as of the latest commit.  How do I make sure I have these changes?  

Many of my problems with compiling date back to old issues I found on this repo.  An example is at https://github.com/jrowberg/i2cdevlib/issues/500.  This concerns defines like prog_ and BUFFER_LENGTH .  Those fix a lot of errors but there seems to be no end to the errors.  I will do a fresh clone and start over and post an issue that lists my problems.  But it would be nice if I knew I had latest fixes.

mark-hahn, 2021-11-24:
Never mind.  I got something simple to work.

For lurkers:  I copied the files in the folder Arduino/i2cdev into my src folder along with the other example source files.  This avoided using any libraries from platformio library collection.  This compiled cleanly.