jrowberg/i2cdevlib #369: Wire.h Bug that causes problems reading DMP form MPU 6050 *possible solution*

matze777, 2018-04-11:
Hi all,

as i just found out that there is a problem with the Wire.h (timeout problem) library when reading the DMP from a MPU6050.

There seems to be no solution so far for this what makes me very sad because i have a great project which now is worthless :-(

Do anyone know if the same problem occurs when you use the MPU9250 instead of the 6050?

Would it may help if you try to use this library:

https://github.com/felias-fogg/SoftI2CMaster

instead of the Wire.h ?? Has someone tried this?

Regards Matthias

jrowberg, 2018-04-11:
While a different or modified I2C master implementation with timeout support is the right solution here, SoftI2CMaster is probably not a good choice since it is bit-banged instead of hardware, which will significantly reduce performance.

I am not sure whether the MPU-9250 exhibits the same issue as the MPU-6050.

bingcheng1998, 2019-05-31:
> While a different or modified I2C master implementation with timeout support is the right solution here, SoftI2CMaster is probably not a good choice since it is bit-banged instead of hardware, which [...]

Hi! I wonder how to use this app in mpu9250. Does it work?
[image]

jrowberg, 2021-09-26:
Closing old issues for cleanup.

closed 2021-09-26 by jrowberg