jrowberg/i2cdevlib #370: Arduino/i2cdev/I2cDev.h should define BUFFER_LENGTH?

themarwhal, 2018-04-13:
Am I missing something here? The directory won't compile because i2cdev.cpp relies on a macro called BUFFER_LENGTH being defined. I assumed it should be defined in I2CDev.h so I defined it in there to make it compile. Is this intentional?

jrowberg, 2018-04-13:
The `BUFFER_LENGTH` constant is defined as `32` in core files for the Arduino Uno and compatible. However, it's not present in all other cores (e.g. ESP32). The I2Cdev library needs to be made more robust across multiple platforms in this regard; an `#ifndef ...` guard with a redefinition should get you by for the moment.

MarquisNelsonManova, 2019-06-10:
hello
how to download this libary
thanks in advance

closed 2020-10-30 by themarwhal