jrowberg/i2cdevlib #376: BUFFER compiling with ESP8266 Arduino

j3nsykes, 2018-05-02:
Hello, 
I have tried adding 
#define BUFFER_LENGTH 32

to my Wire library for the ESP8266 hardware and Arduino hardware but it does not resolve this error when trying to compile code using the i2Cdev and MPU6050 libraries for an Adafruit Huzzah and a standard Node MCU ESP8266. 

This is the compiler error. 
`note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {`

Its worth noting this code compiled and ran fine 1 year ago but now does not compile at all with no changes made. 

Any pointers as to what to try next is much appreciated. 
Thanks

dfvneto, 2018-05-12:
Hey

I had the same issue and I solve it adding #define BUFFER_LENGTH 32 to the i2cdev.h file. I dont know if you are using this file but it may solve your problem!
Good luck my dude

jrowberg, 2021-09-25:
Errors related to undefined `BUFFER_LENGTH` have now been resolved.

closed 2021-09-25 by jrowberg