jrowberg/i2cdevlib #377: Random characters for MPU6050 Examples

ndivuyo, 2018-05-04:
Hey, all of the MPU-6050 example sketches print only random characters (like: ⸮⸮%4%%⸮⸮$⸮!$) they do not print any readable words/characters (not even the init messages). The only sketch that seems to work and print fine is the IMU_ZERO sketch. 

Also I should note that both of the sketches here run perfectly fine for me:
https://playground.arduino.cc/Main/MPU-6050
So I would assume my wiring is correct.

I've tried on both my arduino nano and uno, same results. Do you know what the issue could be? Thank you!

jrowberg, 2018-05-07:
Have you confirmed that the console baud rate is correct (115200 for the DMP6 example sketch, 38400 for the raw example sketch)? That's the first thing I would check. Second for the DMP6 example, make sure that you have uncommented only the desired output mode constant, e.g. `OUTPUT_READABLE_YAWPITCHROLL`. If you have `OUTPUT_TEAPOT` enabled, the reports will be in binary format.

ndivuyo, 2018-05-07:
Hello, thanks for responding. The code has these as the baud rates (unless you mean to confirm the rates through some sort of testing?), also for DMP6 "OUTPUT_READABLE_YAWPITCHROLL" is uncommented. I actually didn't modify the example sketches in any way. Thank you.

jrowberg, 2018-05-07:
The IMU_ZERO sketch starts the host serial interface at 9600 baud. If this works but none of the others appear, then I suspect that your monitoring terminal is set to 9600 baud and you aren't changing it. You need to select a matching baud rate on the host side based on whatever the MCU is using (38400 for the MPU6050_raw sketch and 115200 for the MPU6050_DMP6 sketch).

ndivuyo, 2018-05-07:
Oh I see! I didn't realize that, thank you! As you can tell I am a novice, I appreciate your help and look forward to using your library!