jrowberg/i2cdevlib #378: Using 2 MPU6050

dfvneto, 2018-05-12:
Yo

When using 2 MPU6050 I'm able to get the quaternions (or whatever) via serial but after a few seconds it starts to return FIFO OVERFLOW.
It happens on the arduino uno, ESP8266 and on the ESP32.
Using the Wire.h I can receive the raw data via serial correctly with the 2 mpu's, but I need the quaternions or euler angles to use in Unity. It would help if the generation of the quaternions on the MPU6050_6Axis_MotionApps20.h was explained (if its explained somewhere I would like to apologize because i didnt find it).

I dont know if i'm initializing both MPU6050 correctly, but here's the code:
I erased some comments so it wouldnt be so big
(give me a hint on how to initialize correctly if its wrong please)

#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

MPU6050 mpu1;
MPU6050 mpu2(0x69);

#define LED_PIN 13
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
bool dmpReady2 = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t mpuIntStatus2;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint8_t devStatus2;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t packetSize2;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint16_t fifoCount2;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer
uint8_t fifoBuffer2[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

// orientation/motion vars
Quaternion q2;           // [w, x, y, z]         quaternion container
VectorInt16 aa2;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal2;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld2;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity2;    // [x, y, z]            gravity vector
float euler2[3];         // [psi, theta, phi]    Euler angle container
float ypr2[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

volatile bool mpuInterrupt2 = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady2() {
    mpuInterrupt2 = true;
}

void setup() {
   // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock (200kHz if CPU is 8MHz)
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
[... 163 lines cut ...]
      Serial.print(q.w);
      Serial.print("\t");
      Serial.print(q.x);
      Serial.print("\t");
      Serial.print(q.y);
      Serial.print("\t");
      Serial.println(q.z);
      /*Serial.print("euler\t");
      Serial.print(euler[0] * 180/M_PI);
      Serial.print("\t");
      Serial.print(euler[1] * 180/M_PI);
      Serial.print("\t");
      Serial.println(euler[2] * 180/M_PI);*/

      mpu2.dmpGetQuaternion(&q2, fifoBuffer2);
      Serial.print("quat2\t");
      Serial.print(q2.w);
      Serial.print("\t");
      Serial.print(q2.x);
      Serial.print("\t");
      Serial.print(q2.y);
      Serial.print("\t");
      Serial.println(q2.z);
    }
}

ibrahimseleem, 2019-06-21:
Hello, Do you find any solution to the problem of overflow??

ibrahimseleem, 2019-06-21:
Hello, I found the solution and I tried it right now and it is successfully working.

I used the code in the following link:

https://github.com/Bendgate/Two-MPU6050

But I modify the following

uint8_t fifoBuffer[1024]; // FIFO storage buffer
uint8_t fifoBuffer2[1024]; // (0x69)

Also, I changed all the fifoCount from 1024 to 16384. This is an example:

if ((mpuIntStatus & 0x10) || fifoCount == 16384)