jrowberg/i2cdevlib #380: Integer overflow in uint8_t MPU6050::dmpGetGravity(int16_t*, const uint8_t*)

mikister, 2018-05-25:
While using this library on Arduino I got an error. Here's the error message:

```
C:\Users\...\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGravity(int16_t*, const uint8_t*)':

C:\Users\...\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:685:65: warning: integer overflow in expression [-Woverflow]

         - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384);
```

This can be fixed with the following small change to the code:

>
          - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384L);

IvA101, 2018-07-26:
Why doing that fix the problem??

mikister, 2018-07-27:
Hi, this was a long time ago and I only accidentally ran into this. IIRC this was about a number becoming bigger than the maximum size for its type.
Although I never got to prove if this affected the execution of the code in any way, it still gave error messages, which were kind of annoying. After investigating a bit I found a similar looking piece of code where they also had `2 * 16384` (just to mention at the time I didn't know which piece of code was causing the error since I never used C++ that much) and the suggested solution was changing it to `2 * 16384L` changing the type of that number to a long.
I'm not actively using Arduino or this library now, and back when I posted this it was just a one time activity that I was doing, but I thought to at least post this in case anyone got interested about how to get rid of those error messages.

IvA101, 2018-08-01:
Thanks a lot, it works good for me with this correction "16384L"

ckchiragkhullar, 2018-08-09:
Thanks!! It did work with your correction "16384L"

ghadeerhas, 2019-02-28:
So do you mean we should replace int with long? and what about the uint8?