jrowberg/i2cdevlib #385: i2cdevlib not usable on current arduino esp8266 package

nailujx86, 2018-06-19:
The error that occurs upon compiling with esp8266 package v2.4.1 (all worked fine in v2.3):
```
Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: error: no matching function for call to 'min(uint8_t&, int)'

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note: candidates are:

In file included from c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\julia\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.h:80,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:46:

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^

In file included from c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\julia\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.h:80,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:46:

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^

In file included from c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\julia\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.h:80,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:46:

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^

In file included from c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\julia\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.h:80,

                 from C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:46:

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\julia\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Users\julia\Documents\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^

exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
```

nailujx86, 2018-06-19:
Fixed this in 86a2c29972b90f4684b30493985058f3f0b9d9f6.

nailujx86, 2018-06-19:
Opened a PR: #386

ulno, 2018-09-12:
Is there a reason why nailujx86 PR is not applicable? He is right, this issue makes i2cdevlib unusable on the ESP8266 platform and platformio pulls master in by default. I will see if I can use his fork in platformio instead.

nailujx86, 2018-09-12:
@ulno
It looks like he is not maintaining this anymore. I might make my fork a maintained version with merged pull requests if this doesn't get any attention in the next weeks..

jrowberg, 2018-09-12:
Sorry guys, I have been largely hands-off with this repo for a while due to other responsibilities and the fact that a large portion of the discussion now concerns platforms that I have not personally used. However, I do still see most of the back-and-forth messages.

I'd love to migrate this repo to a structure that makes more sense: organization account, official contributors, CI testing, issue templates, development branches, version tags, and most of all actual **different repos** for different platforms instead of one big monolithic one. My lack of Git/Github knowledge at the beginning of this project set it up for bad maintainability now that it has grown so far beyond where it began.

But I digress. For the moment, I've merged the pull request in question.

closed 2018-10-09 by nailujx86