jrowberg/i2cdevlib #391: MPU6050 - DMP Quaternions with Acceleration Range +-4g

flauth, 2018-08-15:
I assume, that with the current release it is not possible tor read Quaternions from the DMP using a FullScaleAccelRange=1 (means +-4g) , and that changes would include more than just change the divisor in the function (line 617 of MPU6050_6Axis_MotionApps20.h) 
`uint8_t MPU6050::dmpGetQuaternion(Quaternion *q, const uint8_t* packet) `
Is there a way to make these changes easily?

PvdBerg1998, 2018-09-30:
Hey, I'm currently trying to do the same. I changed the hardcoded constants to the value I think they should be but it's giving me garbage results. Did you find a solution?

benjoew, 2018-10-07:
I tried to do the same thing for +-8g without success. I added the line `setFullScaleAccelRange(MPU6050_ACCEL_FS_8);` to function `uint8_t MPU6050::dmpInitialize()` right after the setting of the gyro range. In addition I also adapted the divisors in the other function mentioned.   
The readings for yaw, pitch and roll get totally mixed up and drift a lot. I assume that the solution could be hiding in the dmpMemory that is written to the MPU and that is a problem not that easy to solve.   So, if somebody does have an idea, I also would highly appreciate that.

berkaysaka, 2020-01-28:
I was trying to do similar things like changing the scale ranges for both gyro and accel. Then I found this on the forum. It might worth to dig it a bit more in case of a potential improvement on the newer versions of the firmware.

> DMP sensor fusion works only with gyro at +-2000dps and accel +-2G
> [mllite_test.c, l. 967]

https://www.i2cdevlib.com/forums/topic/221-anyone-experienced-dmp/