jrowberg/i2cdevlib #393: DMP Temperature is not available

ReallyNiceGuy, 2018-08-30:
Hi, 
I noticed that you don't get the temperature from the DMP.
Do you plan to get it? I tried to understand the DMP part, but I could not make heads or tails of it.
The packet is not documented, otherwise I would try myself.
Thanks a lot.

ajuancer, 2018-10-09:
Hi,
I attach you the code. The web and the link to his GitHub are below.
`/*
    MPU6050 Triple Axis Gyroscope & Accelerometer. Temperature Example.
    Read more: http://www.jarzebski.pl/arduino/czujniki-i-sensory/3-osiowy-zyroskop-i-akcelerometr-mpu6050.html
    GIT: https://github.com/jarzebski/Arduino-MPU6050
    Web: http://www.jarzebski.pl
    (c) 2014 by Korneliusz Jarzebski
*/

#include <Wire.h>
#include <MPU6050.h>

MPU6050 mpu;

void setup() 
{
  Serial.begin(115200);

  Serial.println("Initialize MPU6050");

  while(!mpu.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G))
  {
    Serial.println("Could not find a valid MPU6050 sensor, check wiring!");
    delay(500);
  }
}

void loop()
{
  float temp = mpu.readTemperature();

  Serial.print(" Temp = ");
  Serial.print(temp);
  Serial.println(" *C");
  
  delay(500);
}`

ReallyNiceGuy, 2018-10-09:
Thank you for your response.
I don't think this will help me very much, as I was talking about the DMP. I don't want to have to do another read from the I2C to get this information.
Also, I forgot to mention that I am using the ESP_IDF implementation, not the Arduino.