jrowberg/i2cdevlib #394: 16 bit overflow in MPU6050_6Axis_MotionApps20.h

harbaum, 2018-09-03:
2 * 16384 will evaluate to -32767 on a machine that uses signed 16 bit for integers in line:

https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050_6Axis_MotionApps20.h#L685

jrowberg, 2018-09-03:
Does an 8-bit ATMega do this, or some other platform? Or is it just a general warning for potentially non-portable code?

harbaum, 2018-09-03:
Happens on avr8.

micbuh, 2019-01-11:
it's a avr8 warning:
`In file included from src/main.cpp:49:0:
lib/MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGravity(int16_t*, const uint8_t*)':
lib/MPU6050/MPU6050_6Axis_MotionApps20.h:689:65: warning: integer overflow in expression[-Woverflow]
- (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384);
^`

followed by:
`lib/MPU6050/MPU6050.h:436:7: warning: type 'struct MPU6050' violates one definition rule[-Wodr]
class MPU6050 {
^

lib/MPU6050/MPU6050.h:436:7: note: a different type is defined in another translation unit
class MPU6050 {
^

lib/MPU6050/MPU6050.h:1026:18: note: the first difference of corresponding definitions is field 'dmpPacketBuffer'
uint8_t *dmpPacketBuffer;
^

lib/MPU6050/MPU6050.h:436:7: note: a type with different number of fields is defined in another translation unit
class MPU6050 {
^`

JAY-M-ERICSSON, 2020-03-06:
just change the (2 * 16384) into (2 * 16384L) and that will fix it for all platforms.

jrowberg, 2021-09-26:
Overflow warning fixed with recent commits, and the ODR problem will be corrected in the next week or so.

closed 2021-09-26 by jrowberg