jrowberg/i2cdevlib #397: MPU 6050 Library Orientation

ThePoshAI, 2018-09-21:
For my application of the sensor, a VR headset, I need the axes to be redefined to Yaw about the x-axis, Pitch about the y-axis and roll about the z-axis. Looking at example code and at the MPU6050_9Axis_MotionApps41.h library, I found that to redefine the axis the library must be modified. Shown below I have modified dmpGetYawPitchRoll to use the different axis so that the yaw and pitch now work correctly. However, the roll does not work correctly.

Modified code:
`
uint8_t MPU6050::dmpGetYawPitchRoll(float *data, Quaternion *q, VectorFloat *gravity) {
    // yaw: (about x axis)
    data[0] = atan2(2*q -> y*q -> z - 2*q -> w*q -> x, 2*q -> w*q -> w + 2*q -> y*q -> y - 1);
    // pitch: (nose up/down, about y axis)
    data[1] = atan(gravity -> y / sqrt(gravity -> z*gravity -> z + gravity -> x*gravity -> x));
    // roll: (tilt left/right, about z axis)
    data[2] = atan(gravity -> z / sqrt(gravity -> y*gravity -> y + gravity -> x*gravity -> x));
    return 0;
}
`
original:
`
uint8_t MPU6050::dmpGetYawPitchRoll(float *data, Quaternion *q, VectorFloat *gravity) {
    // yaw: (about Z axis)
    data[0] = atan2(2*q -> x*q -> y - 2*q -> w*q -> z, 2*q -> w*q -> w + 2*q -> x*q -> x - 1);
    // pitch: (nose up/down, about Y axis)
    data[1] = atan(gravity -> x / sqrt(gravity -> y*gravity -> y + gravity -> z*gravity -> z));
    // roll: (tilt left/right, about X axis)
    data[2] = atan(gravity -> y / sqrt(gravity -> x*gravity -> x + gravity -> z*gravity -> z));
    return 0;
}
`

When facing forward the roll works correctly, turning 90 degrees clockwise or anti-clockwise disables all roll and turning 180(from forwards) causing the roll to become inverted.
I believe my issue may stem from not modifying quaternation or gravity to the new axis but I am lost when trying to modify those sections of the code.

Looking at the Issues page I have seen people posting about issues that they think are of calibration but share symptoms of incorrect axis mapping I have experienced.

Regardless, thank you for the great work.

ThePoshTux, 2018-09-23:
#64

ThePoshTux, 2018-09-23:
#64