jrowberg/i2cdevlib #399: MPU6050_DMP6/MPU6050_DMP6.ino  does not compile for the M0

dsyleixa, 2018-09-27:
I tried to compile the example https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino 

but it does not compile, It throws the error

> Bibliothek I2Cdev im Ordner: D:\Arduino\portable\libraries\I2Cdev (legacy) wird verwendet
> Bibliothek Wire in Version 1.0 im Ordner: D:\arduino\portable\packages\adafruit\hardware\samd\1.2.3\libraries\Wire  wird verwendet
> Bibliothek MPU6050 im Ordner: D:\Arduino\portable\libraries\MPU6050 (legacy) wird verwendet
> exit status 1
> Fehler beim Kompilieren für das Board Adafruit ItsyBitsy M0.

dsyleixa, 2018-10-08:
no idea about a fix?

dsyleixa, 2018-10-11:
any suggestions about a fix to this issue?

dsyleixa, 2018-10-15:
why is there no suggestion how to fix this issue?

jrowberg, 2018-10-15:
Fixed by in https://github.com/jrowberg/i2cdevlib/commit/900b8f959e9fa5c3126e0301f8a61d45a4ea99cc:

https://github.com/jrowberg/i2cdevlib/blob/900b8f959e9fa5c3126e0301f8a61d45a4ea99cc/Arduino/I2Cdev/I2Cdev.cpp#L90-L93

closed 2018-10-15 by jrowberg

dsyleixa, 2018-10-15:
thank you, now it compiles !

Nonetheless, it does not work longer than 30sec, then there is a runtime error, program stops:

ypr	74.65	-12.55	9.63
ypr	74.60	-12.41	9.58
ypr	74.57	-12.36	9.58
ypr	74.52	-12.38	9.64
ypr	74.43	-12.46	9.71
ypr	74.19	-12.59	9.80
ypr	73.76	-12.80	10.05
ypr	73.24	-12.94	10.35
ypr	72.68	-12.93	10.59
ypr	72.08	-12.77	10.83
ypr	71.46	-12.47	11.08
ypr	70.82	-12.15	11.29
ypr	70.18	-11.85	11.48
ypr	69.54	-11.56	11.69
ypr	68.89	-11.29	11.93
FIFO overflow!
ypr	49.25	1.73	15.39
ypr	48.20	2.31	15.72
ypr	47.27	2.92	15.99
ypr	46.50	3.35	16.27

dsyleixa, 2018-10-15:
this is my current code:

```
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
// https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino 
//
// Changelog:
//      2016-04-18 - Eliminated a potential infinite loop
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2012 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

#ifndef _BV
#define _BV(n) (1<<(n))
#endif

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
/*
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif
*/
// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
[... 289 lines cut ...]
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

```