jrowberg/i2cdevlib #400: how to download and install just  a single lib (MPU6050) ?

dsyleixa, 2018-10-07:
how to download and install just  a single lib (MPU6050) ?

ajuancer, 2018-10-07:
I have done it, I attach it [here](https://drive.google.com/open?id=1YKh6de9f3neBtFuB4ZeSCPXde---QrpY). 
You only have to copy the files you want.
Remember to move the folder I attach to Documents/Arduino/Libraries

dsyleixa, 2018-10-08:
yes, I know how to install the entire i2cdevlib, but I just wanted 1 single lib (MPU6050).
Of course I can copy and paste them from the entire zip file arbitrarily but I asumed there was a simpler way for either single lib.

jrowberg, 2018-10-08:
This is an unfortunate side-effect of the way this project has grown into a multi-platform beast all under a single monolithic repository. I am working on a significant "reboot" that will be much easier for community development and platform-specific/library-specific integration in modern ecosystems like the Arduino library manager and PlatformIO, etc. But there's no good way to do the same thing with the existing single i2devlib repo without breaking a whole lot of backwards compatibility. :(

closed 2018-10-08 by jrowberg

dsyleixa, 2018-10-08:
thanks for your reply! I appreciate very much your efforts on providing all that stuff, I just didn't understand from the start  that all is in the big repo .zip and can be arbitrarily inalled by c+p'ing the single folders.
That's actually not an issue now any more, thanks again!

ajuancer, 2018-10-08:
I hope it was worth it!!, @dsyleixa 
If you have any questions, feel free to write here.

dsyleixa, 2018-10-09:
> 
> 
> I hope it was worth it!!, @dsyleixa
> If you have any questions, feel free to write here.

thanks, tbh, there is still a major unresolved issue reported here:
https://github.com/jrowberg/i2cdevlib/issues/399