jrowberg/i2cdevlib #408: FIFO overflow!

ajuancer, 2018-10-14:
Hi, after solving some problems i've been posting here, now i'm reading FIFO overflow in the serial monitor. As i have read the example code,  i know that the code it very inefficient, so I decide to add `mpu.resetFIFO();` after the line of code that i write, but the problem continues. I don't know what to do, someone know?
(here is the code i aded)
`    if (ypr[1]* 180/M_PI >= 5.4) {
      Serial.println("NICE");
      printStringWithShift(mensaje1, 40);
      mpu.resetFIFO();
    }`

dsyleixa, 2018-10-15:
same to me:
ypr	74.65	-12.55	9.63
ypr	74.60	-12.41	9.58
ypr	74.57	-12.36	9.58
ypr	74.52	-12.38	9.64
ypr	74.43	-12.46	9.71
ypr	74.19	-12.59	9.80
ypr	73.76	-12.80	10.05
ypr	73.24	-12.94	10.35
ypr	72.68	-12.93	10.59
ypr	72.08	-12.77	10.83
ypr	71.46	-12.47	11.08
ypr	70.82	-12.15	11.29
ypr	70.18	-11.85	11.48
ypr	69.54	-11.56	11.69
ypr	68.89	-11.29	11.93
FIFO overflow!
ypr	49.25	1.73	15.39
ypr	48.20	2.31	15.72
ypr	47.27	2.92	15.99
ypr	46.50	3.35	16.27
ypr	45.88	3.68	16.48
ypr	45.30	3.98	16.57
ypr	44.57	4.38	16.81
ypr	43.99	4.66	16.82
ypr	43.22	4.84	16.93
ypr	42.32	5.02	16.97

dsyleixa, 2018-10-15:
this is my current code:

```
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
// https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino 
//
// Changelog:
//      2016-04-18 - Eliminated a potential infinite loop
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2012 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

#ifndef _BV
#define _BV(n) (1<<(n))
#endif

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
/*
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif
*/
// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
[... 289 lines cut ...]
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

```

ajuancer, 2018-10-16:
I think the only one who can truly help us Mr. Rowberg (@jrowberg )

dsyleixa, 2018-10-18:
@dev:
will there soon be a FIFO overflow fix?

dsyleixa, 2018-10-20:
PS:
is the FIFO overflow caused by the i2cdevlib source code or by the MPU6050 hardware?

ajuancer, 2018-10-20:
I think it's caused by the source coee

dsyleixa, 2018-10-20:
@jrowberg - what do you think?

jrowberg, 2018-10-20:
The FIFO overflow condition occurs when the IMU reports (via interrupt register read) that the FIFO has overflowed--note the use of `MPU6050_INTERRUPT_FIFO_OFLOW_BIT` in one of the `if` conditions. FIFO queuing is handled entirely by the IMU; it's not an application-level buffer.

When the FIFO overflows, fundamentally this means that the application cannot read data as fast as the IMU is generating it. Often this is caused by the CPU having too many other things to do between reads. It may be that the I2C data rate is too slow (100 kHz never helps, if the I2C clock is still set there). Monitoring the I2C data lines with a logic analyzer is the easiest way to see how often the reads are occurring and if they are using a lot of the CPU's processing time. Serial output is also often very slow, though this is less of a problem when using USB CDC for host output instead of hardware serial at slow baud rates. It is hard to say where other slow execution points may occur without detailed profiling tools.

You can reduce the DMP output frequency by increasing the rate divisor value inside the `..._MotionApps20.h` file:

https://github.com/jrowberg/i2cdevlib/blob/900b8f959e9fa5c3126e0301f8a61d45a4ea99cc/Arduino/MPU6050/MPU6050_6Axis_MotionApps20.h#L272-L274

The data will be generated more slowly, but it will allow the CPU to keep up.

dsyleixa, 2018-10-20:
I actually doubt that the CPU is too slow, because 
- I am just running the DMP7 example, nocthing else,
- I am running the i2c bus at 400 kHz as it's the original code
- I am using a ARM Cortex M4 board at 120MHz cpu clock which is ~ 8x faster than AVR clock.

So if it does not get stuck on AVRs by FIFO overflow: why should it do that on an M4 ?
Perhaps the FIFO thing is not the actual reason why the program hangs up after some seonds on my M4 as already assumed by me?

jrowberg, 2018-10-20:
Can you check the I2C lines with a logic analyzer?

dsyleixa, 2018-10-20:
no, I don't have such a thing

jrowberg, 2018-10-20:
Can you check the voltage levels with a multimeter? If either is stuck at GND after the hang occurs, then the MPU is misbehaving (which is an oft-reported issue, see #252).

jrowberg, 2018-10-20:
This library takes the raw accel/gyro data as inputs; to use it then, you must have the MPU-6050 operating in a very different mode (DMP inactive), so it is less likely that it will misbehave. In fact, I don’t recall ever seeing any reports of the IMU freezing in raw data mode.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dsyleixa, 2018-10-20:
2nd intermediate result:

I had to migrate to another PC (a Intel i5, the former one was an older and slower core2duo E2660).
On this new one it currently runs fine during about 5 min without any FIFO issues on Serial monitor, strangely.

I'll see if I can reproduce the FIFO thing on the old one when I'll be back there and then will report about voltage levels in case it hangs up again.

dsyleixa, 2018-10-21:
Update:
this "freezing" error showing intermediate FIFO overflow on Serial console is always reproducable just at my old core2duo (Win7-64 pro, Serial.begin(155200), Arduino portable 1.8.5 ), attached to either PC USB port. 
When freezing there, both SDA and SCL have 3.28V voltage level.

FTM, not reproducable though on my new PC (also Win7-64 pro, Serial.begin(155200), identical Arduino portable 1.8.5 ).

ajuancer, 2018-10-26:
@jrowberg What do I have to add on line 272?

jrowberg, 2018-10-26:
@ajuancer You don’t have to add anything; just increase the divisor value from 0x01 to something bigger, like 0x02 or 0x03. The larger this value is, the slower the DMP will generate data.

vicoz2009, 2018-10-27:
This is an interesting thread.  I did experience overflow before and might check out line 272...wherever that is.

ajuancer, 2018-10-28:
Nothing... I still have the same error, don't know what to do

ajuancer, 2018-10-28:
@jrowberg Nothing... I still have the same error, don't know what to do

ajuancer, 2018-11-04:
All the code I have write is here: https://github.com/ajuancer/smart_bicycle/blob/master/move-matrix.ino

john-bessire, 2019-01-01:
The FIFO overflow problem finally went away with setting  #define MPU6050_DMP_FIFO_RATE_DIVISOR 0x04 but they still is an occasionally lock up for several seconds before data resumes.

wuweijia1994, 2019-04-25:
Just curious about when printing out the "FIFO Overflow" message.

 Why the condition is ((mpuIntStatus & _BV(MPU6050_INTERRUPT_FIFO_OFLOW_BIT)) || fifoCount >= 1024)? 

Because "MPU6050_INTERRUPT_FIFO_OFLOW_BIT" is defined to be 4 and "_BV(MPU6050_INTERRUPT_FIFO_OFLOW_BIT)" is always 16. 

So the condition equals to (mpuIntStatus || fifoCount >= 1024) which does not make sense.

madusudananr, 2019-11-04:
Interestingly, this happens on my breadboard setup (nano + MPU6050) too ...when the pitch or the roll is beyond a small initial value.

And more interestingly, if i touch wire insulation that connects INT on the MPU to the arduino interrupt pin - this problem does not happen !!

Need to explore this !

jcrosbyME, 2019-11-04:
Try putting a very small capacitor (5 to 10 pf) between the INT signal line and ground. This will shunt any hi frequency EMI to ground.

madusudananr, 2019-11-05:
That was the first instinct - but I put a larger capacitor - which obviouslly wouldn't have worked - dumb of me.  Will try the small cap.  

Also the pullup resistor worked, although not sure if it is the right thing to do.

elpidiovaldez, 2019-12-15:
I have been through hell over the last week with intermittent hangs and bad data in a deployed circuit (although everything worked perfectly on a small breadboard).  I now firmly believe that nearly all the unreliability issues experienced by many people are due to electrical problems, not the I2CDev library or the MPU6050 library.  The problems began for me when the connection between the Arduino and the MPU6050 was longer than 8-10cm (not very long really !).

The solution is so simple.  Add a SERIES resistor in the SDL and SDA connections (near the Arduino).  Resistance should be approx 30 - 200 Ohm (100 Ohm worked for me).  The resistor damps 'ringing' on the lines and it is a known practice in electronic engineering forums. This is NOT THE PULLUP RESISTOR - it serves a different function.  There can of course be other electrical problems but this is a great starting point.

I appreciate that this is a software discussion, but I think many people will save themselves some pain by trying this.  If  I2C communication is reliable then the timeout problem in i2cDev becomes secondary, since it is never needed.

msaine, 2019-12-15:
To All,

I agree! From what I have been reading on this forum the problem is electrical, not software. Especially when a bread board works with no motors involved and starts to behave badly when motors get involved.

I also use teensy processors which eliminates some of the 5 vs 3.3 problems. I realize that solution is not for everyone but having an Arm processor clocked at 96Mhz does help 😊

Again, my purpose is to make others think, not step on anyone’s toes. The power of these forums is getting many people to think and contribute not to have the “best answer”.

Series resistors on the SDL/SDA lines helps greatly but the Interrupt problem requires another approach but I am not positive what is the best method. I am seriously considering putting my processor and 6050 in a shielded can to help shield from rfi/efi, and adding shielding and capacitors to the power leads on the motors.

FWIW many project managers these days want to “fix it in software, we can’t change the hardware”. Unless it is a combined approach the bugs remain. So do we have a true electrical engineer out there that has experience in RFI shielding and prevention???

msaine

jcrosbyME, 2019-12-16:
Hi,

You might consider a 10pF to 20 pF capacitor from the interrupt line to ground to form a low pass filter. This will effectively shunt the typically higher frequency noise to ground.  From what I have seen, this is a fairly common remedy to noise on a digital line. This value of the capacitor can be increased, but at some point it will act as a delay to the interrupt or distort the leading and trailing edges of the signal to a slope. If this works it explains why the breadboard worked when the final circuit didn’t as there is parasitic capacitance in the breadboard.

                -Jay Crosby

ralphs99, 2019-12-16:
Hi all,

It may be that the series resistors mentioned by Msaine are not actually working as series termination resistors but instead are blocking noise from being conducted down the I2C wires. This can be tested by replacing the series resistors with ferrite beads with a resistance of several hundred ohms at low frequencies in the MHz range eg Bourns Inc. FB20022-4B-RC. 

While not offering anything over a simple series resistor on I2C interconnects, ferrite beads may also help on interrupt (in addition to the small capacitors suggested by Jay Crosby) and power wires where the power loss from a resistor cannot be tolerated.

Ferrite beads of this type are not really a production solution, but they may help in determining where noise is coming from and whether the problem is one of radiated or conducted noise.

Cheers, Ralph

elpidiovaldez, 2019-12-17:
Series resistors on the SCL and SDA lines made a huge difference to the reliability for me on a bread board implementatation, but eventually the system hung again.  I tried putting a series resistor into the INT line too on the grounds that it might also suffer from ringing.  That made the breadboard implementation perfectly reliable.  Unfortunately when I put the series resistors on the circuit board, in the robot, the system was still unreliable (to avoid doubt there were no motors or PWM in use during the testing).  It beats me - the wires were shorter in this implementation than on the breadboard.  Does anybody know if there should be a difference due to the resistors being at the Arduino side vs. the sensor side of the connections ?

I cannot believe how difficult it is to use this sensor.  I have spent a full week just building and rebuilding the circuit.