jrowberg/i2cdevlib #410: after pressing reset button: program hangs up, serial monitor freezes

dsyleixa, 2018-10-15:
after pressing reset button: program hangs up, serial monitor freezes showing jus this message:

`Initializing I2C devices...

(Adafruit Feather M4)
`

ThePoshTux, 2018-10-31:
I do not know about how to correct this issue but I do know the intended method for resetting. Next time you want to reset the MPU/Arduino. Sending a singular ``R`` into the serial terminal and that should reset the board and MPU correctly.
Anyway, I believe the issue is caused by the MPU not being reset when the Arduino is reset as the MPU neither receives commands to stop and then start or reset upon power disconnection, leaving the MPU in an incorrect state.

ErikBob, 2018-11-25:
> Anyway, I believe the issue is caused by the MPU not being reset when the Arduino is reset as the MPU neither receives commands to stop and then start or reset upon power disconnection, leaving the MPU in an incorrect state.

MPU6050_DMP6 includes a call to a function that dumps a whole memory block into the MPU on startup since its memory seems to be volatile.
If it is indeed volatile how could it be left in an incorrect state?

jrowberg, 2018-11-25:
The MPU's volatile memory will remain intact if the host MCU resets but the whole system remains powered. If there's some kind of state machine problem inside the MPU, it may not accept the new DMP microcode dump correctly. It's not a bad idea to specifically reset the MPU on boot, either via hardware (if connected) or software.