jrowberg/i2cdevlib #413: dmp6 example hangs up and gets stuck after a couple of seconds (randomly)

dsyleixa, 2018-10-20:
the dmp6 example hangs up and gets stuck after a couple of seconds (randomly).

Perhaps it's caused by FIFO stack overflow as reportes here
https://github.com/jrowberg/i2cdevlib/issues/408#issue-369917664 ,
 but I am not sure about the actual reason, perhaps it's due to different faults or errors,  because the Serial output is not hte last message befor it gets stuck:

(IDE 1.8.5, Adafruit Feather M4)

ypr 74.65 -12.55 9.63
ypr 74.60 -12.41 9.58
ypr 74.57 -12.36 9.58
ypr 74.52 -12.38 9.64
ypr 74.43 -12.46 9.71
ypr 74.19 -12.59 9.80
ypr 73.76 -12.80 10.05
ypr 73.24 -12.94 10.35
ypr 72.68 -12.93 10.59
ypr 72.08 -12.77 10.83
ypr 71.46 -12.47 11.08
ypr 70.82 -12.15 11.29
ypr 70.18 -11.85 11.48
ypr 69.54 -11.56 11.69
ypr 68.89 -11.29 11.93
FIFO overflow!
ypr 49.25 1.73 15.39
ypr 48.20 2.31 15.72
ypr 47.27 2.92 15.99
ypr 46.50 3.35 16.27
ypr 45.88 3.68 16.48
ypr 45.30 3.98 16.57
ypr 44.57 4.38 16.81
ypr 43.99 4.66 16.82
ypr 43.22 4.84 16.93
ypr 42.32 5.02 16.97

^~~~~~~~~~~~~~~^
after this last line the program freezes completely, no Serial output any more at all.
Pressing reset button does not the restart the program correctly then.

dsyleixa, 2018-10-20:
this is my current code:
(IDE 1.8.5, Adafruit Feather M4)

// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
// https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino 
//
// Changelog:
//      2016-04-18 - Eliminated a potential infinite loop
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2012 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

#ifndef _BV
#define _BV(n) (1<<(n))
#endif

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
/*
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif
*/
// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
[... 287 lines cut ...]
            Serial.print("\t");
            Serial.print(aaWorld.y);
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

dsyleixa, 2018-10-20:
as stated, I actually doubt that it's a duplicate of   https://github.com/jrowberg/i2cdevlib/issues/408

**_perhaps it's due to different faults or errors, because the Serial output message

    FIFO overflow!

is not the ultimate last message before it gets stuck!_**

jrowberg, 2018-10-20:
Do you have a logic analyzer to observe the I2C states? Are the SCL and SDA lines returning to idle (logic HIGH) or are either of them stuck low?

dsyleixa, 2018-10-20:
no, I don't have such a thing.

1st intermediate result:
this lib works fine, never hangs up
https://github.com/TKJElectronics/KalmanFilter
unfortunately it dont provide yaw as needed.

dsyleixa, 2018-10-21:
Update:
this "freezing" error showing intermediate FIFO overflow on Serial console is always reproducable just at my old core2duo (Win7-64 pro, Serial.begin(155200), Arduino portable 1.8.5 ), attached to either PC USB port.

FTM, not reproducable though on my new PC (also Win7-64 pro, Serial.begin(155200), identical Arduino portable 1.8.5 ).

closed 2018-10-21 by dsyleixa