jrowberg/i2cdevlib #417: Serial Monitor

Natalieeeeeeeeee, 2018-11-01:
Hi Mr Rowberg,

I'm trying to upload the MPU6050_DMP6 folder on to my computer and I've managed to do so. However, the serial monitor always fails to connect and display the data even though I have followed all the instructions on your website

https://www.instructables.com/id/MPU6050-Arduino-6-Axis-Accelerometer-Gyro-GY-521-B/

Could you please give me suggestions on what could be going wrong and how I can fix the problem so that my serial monitor can display stuff? Thanks!

ThePoshTux, 2018-11-01:
Hello, 1st welcome to GitHub.
It would be easier for me and others to help you if you state the exact setup you are using and what you have tried so far. I gauge that you are in fact using an Arduino Uno? Is the code you have uploaded "MPU6050_DMP6"? Finally, are you trying to view the serial data through the Arduino ide or through another application? You may have the issue of having two applications on your computer trying to access the virtual serial port that all Arduino boards use.

Natalieeeeeeeeee, 2018-11-02:
Thanks for replying!

I'm using a Maker Uno
The code I'm using is MPU6050_DMP6
I'm trying to view the serial data through the Arduino ide.

My problem is different now though. I've managed to connect up the accelerometer and the Arduino. However, there's always an error uploading the sketch even though I already added the 12Cdev and MPU6050 files to the Arduino library.

ThePoshTux, 2018-11-02:
Ah okay, Is the issue during the compiling stage or the uploading stage?
If it's the uploading stage, have you tried navigating to tools and then checking if the board and port are set correctly? If it's the compiling stage could you copy what the error message is?