jrowberg/i2cdevlib #419: MPU6050 Accuracy vs Digital Protractor

erefour1D, 2018-12-05:
Hope all is well. My name is Eric and I've been working with different imu sensors, specifically the MPU6050 GY521 breakout board. I've developed a simple complementary filter type algorithm (based of several online libraries) to compute the roll and pitch angular values. Initially, I thought everything was working properly until I did an accuracy test using a digital protractor. I noticed that each sensor has a unique offset and in some cases, the offset would change values over a range of motion of about 90 degrees. For example, at 0 degree, the sensor may be off by -1 degree, but at 90 degrees, it would become off by +5 degrees. 

I thought this issue may have been resulting from my algorithm, so I decided to use Jeff Rowberg's MPU6050 library and noticed the same behavior. So far, I've tested multiple MPU6050s (from different vendors) and still see this non-constant offset issue.

I was wondering if anyone else ever noticed this inconsistent behavior when working with the MPU6050?

Thanks

ErikBob, 2018-12-08:
I can't reproduce the error.

What do you mean exactly by offset? Seems wierd to give an offset in degree if the sensor is measuring angular momentum.

erefour1D, 2018-12-08:
Hey,

Originally when I said offset, I was referring to the error in the angles values when I compare the computed pitch/roll angles (in degrees) to the true angular reading of a digital protractor. For instance, if I have the MPU6050 fixed to a breadboard and then rotate that breadboard to 45 degrees, the computed pitch/roll angle may show 43 degrees, giving an error of 2 degrees. Now if I rotate it to 90 degrees, the computed pitch/roll angle may now read 95 degrees, which shows that the "error" changed from being 2 degrees off to now 5 degrees.

I ended up getting  it to work properly using Jeff's library over the past few of days once I took into account the "bias" in the gyro and accelerometer raw values at rest. However, I'm still not able to get it to work with a complementary filter. Probably something else I'm not doing correctly. 

Hope this clears up any confusion.