jrowberg/i2cdevlib #421: If I include "MPU6050_6Axis_MotionApps20.h" into my Project I get compilation errors

Universumgames, 2018-12-25:
Hey,
I have a problem with the "MPU6050_6Axis_MotionApps20.h" file.
After including this file to my project I get a huge amount of error messages:

.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpInitialize()':
MPU.cpp:(.text._ZN7MPU605013dmpInitializeEv+0x0): multiple definition of `MPU6050::dmpInitialize()'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605013dmpInitializeEv+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpPacketAvailable()':
MPU.cpp:(.text._ZN7MPU605018dmpPacketAvailableEv+0x0): multiple definition of `MPU6050::dmpPacketAvailable()'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605018dmpPacketAvailableEv+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetAccel(int*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605011dmpGetAccelEPiPKh+0x0): multiple definition of `MPU6050::dmpGetAccel(int*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605011dmpGetAccelEPiPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetAccel(short*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605011dmpGetAccelEPsPKh+0x0): multiple definition of `MPU6050::dmpGetAccel(short*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605011dmpGetAccelEPsPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetAccel(VectorInt16*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605011dmpGetAccelEP11VectorInt16PKh+0x0): multiple definition of `MPU6050::dmpGetAccel(VectorInt16*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605011dmpGetAccelEP11VectorInt16PKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetQuaternion(int*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605016dmpGetQuaternionEPiPKh+0x0): multiple definition of `MPU6050::dmpGetQuaternion(int*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605016dmpGetQuaternionEPiPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetQuaternion(short*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605016dmpGetQuaternionEPsPKh+0x0): multiple definition of `MPU6050::dmpGetQuaternion(short*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605016dmpGetQuaternionEPsPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetQuaternion(Quaternion*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605016dmpGetQuaternionEP10QuaternionPKh+0x0): multiple definition of `MPU6050::dmpGetQuaternion(Quaternion*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605016dmpGetQuaternionEP10QuaternionPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetGyro(int*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605010dmpGetGyroEPiPKh+0x0): multiple definition of `MPU6050::dmpGetGyro(int*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605010dmpGetGyroEPiPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetGyro(short*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605010dmpGetGyroEPsPKh+0x0): multiple definition of `MPU6050::dmpGetGyro(short*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605010dmpGetGyroEPsPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetGyro(VectorInt16*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605010dmpGetGyroEP11VectorInt16PKh+0x0): multiple definition of `MPU6050::dmpGetGyro(VectorInt16*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605010dmpGetGyroEP11VectorInt16PKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetLinearAccel(VectorInt16*, VectorInt16*, VectorFloat*)':
MPU.cpp:(.text._ZN7MPU605017dmpGetLinearAccelEP11VectorInt16S1_P11VectorFloat+0x0): multiple definition of `MPU6050::dmpGetLinearAccel(VectorInt16*, VectorInt16*, VectorFloat*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605017dmpGetLinearAccelEP11VectorInt16S1_P11VectorFloat+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetLinearAccelInWorld(VectorInt16*, VectorInt16*, Quaternion*)':
MPU.cpp:(.text._ZN7MPU605024dmpGetLinearAccelInWorldEP11VectorInt16S1_P10Quaternion+0x0): multiple definition of `MPU6050::dmpGetLinearAccelInWorld(VectorInt16*, VectorInt16*, Quaternion*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605024dmpGetLinearAccelInWorldEP11VectorInt16S1_P10Quaternion+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetGravity(short*, unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605013dmpGetGravityEPsPKh+0x0): multiple definition of `MPU6050::dmpGetGravity(short*, unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605013dmpGetGravityEPsPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetGravity(VectorFloat*, Quaternion*)':
MPU.cpp:(.text._ZN7MPU605013dmpGetGravityEP11VectorFloatP10Quaternion+0x0): multiple definition of `MPU6050::dmpGetGravity(VectorFloat*, Quaternion*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605013dmpGetGravityEP11VectorFloatP10Quaternion+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetEuler(float*, Quaternion*)':
MPU.cpp:(.text._ZN7MPU605011dmpGetEulerEPfP10Quaternion+0x0): multiple definition of `MPU6050::dmpGetEuler(float*, Quaternion*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605011dmpGetEulerEPfP10Quaternion+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetYawPitchRoll(float*, Quaternion*, VectorFloat*)':
MPU.cpp:(.text._ZN7MPU605018dmpGetYawPitchRollEPfP10QuaternionP11VectorFloat+0x0): multiple definition of `MPU6050::dmpGetYawPitchRoll(float*, Quaternion*, VectorFloat*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605018dmpGetYawPitchRollEPfP10QuaternionP11VectorFloat+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpProcessFIFOPacket(unsigned char const*)':
MPU.cpp:(.text._ZN7MPU605020dmpProcessFIFOPacketEPKh+0x0): multiple definition of `MPU6050::dmpProcessFIFOPacket(unsigned char const*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605020dmpProcessFIFOPacketEPKh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpReadAndProcessFIFOPacket(unsigned char, unsigned char*)':
MPU.cpp:(.text._ZN7MPU605027dmpReadAndProcessFIFOPacketEhPh+0x0): multiple definition of `MPU6050::dmpReadAndProcessFIFOPacket(unsigned char, unsigned char*)'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605027dmpReadAndProcessFIFOPacketEhPh+0x0): first defined here
.pioenvs\esp32dev\src\MPU.cpp.o: In function `MPU6050::dmpGetFIFOPacketSize()':
MPU.cpp:(.text._ZN7MPU605020dmpGetFIFOPacketSizeEv+0x0): multiple definition of `MPU6050::dmpGetFIFOPacketSize()'
.pioenvs\esp32dev\src\Handmotioncontroller_Arduino.ino.cpp.o:Handmotioncontroller_Arduino.ino.cpp:(.text._ZN7MPU605020dmpGetFIFOPacketSizeEv+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp32dev\firmware.elf] Error 1

It would be great if my problem could be solved.
And by the way this error only occurs if I INCLUDE the "MPU6050_6Axis_MotionApps20.h" and not with the "MPU6050.h"....

DerekLF, 2019-01-01:
It looks like you've included the library multiple times

could you post your code.

gparvind, 2019-02-03:
I believe it's because the function and variable definitions are also in the header file. They should be moved to cpp file. Meanwhile, If you include this in another cpp file (MPU.cpp) and not in header file (MPU.h?) it shoud be ok. This is what I observed in my project.

carolambek, 2019-10-08:
Got the same issue and solved it by adding all function definitions from MPU6050_6Axis_MotionApps20.h into MPU6050.cpp and just left in MPU6050_6Axis_MotionApps20.h the variable declarations and all the #define. The declarations of functions originally defined in MPU6050_6Axis_MotionApps20.h are actually already in MPU6050.h when MPU6050_INCLUDE_DMP_MOTIONAPPS20 is defined (which is done in MPU6050_6Axis_MotionApps20.h).

closed 2022-06-30 by Universumgames

divStar, 2023-07-31:
Honestly, this is closed, but the issue persists. There should be a PR to fix `MPU6050_6Axis_MotionApps20.h`. If I am allowed to, I could probably make one myself (though mind you I am a C++ newbie, but not a newbie to software development).

For me it's if I dare to import the `MPU6050_6Axis_MotionApps20.h` anywhere but the `main.h` / `main.cpp`, I get build errors. But since I have a dedicated class for managing the 6-axis controller, I would like to move everything there.

Unfortunately I cannot 😞 .

ZHomeSlice, 2023-07-31:
Note that version 2.0 of the embedded DMP firmware has no documentation.  Version 6.12 is the latest I would recommend using this one. 
Z