jrowberg/i2cdevlib #424: Violation of one definition rule

paynterf, 2019-01-17:
When I compile my project using your wonderful i2cdevlib, I get the following output

`Compiling debug version of 'FourWD_WallE2_V1' for 'Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)'
MPU6050.h: 401:7: warning: type 'struct MPU6050' violates one definition rule [-Wodr]
   class MPU6050 {
MPU6050.h:401: note  a different type is defined in another translation unit
   class MPU6050 {
MPU6050.h:781: note  the first difference of corresponding definitions is field dmpPacketBuffer
   uint8_t *dmpPacketBuffer
MPU6050.h:983: note  a field with different name is defined in another translation unit
   uint8_t devAddr
`

I know I can suppress these errors by adding '-Wodr' to my compiler switches, but this doesn't seem like a good idea.  Any idea why this is happening?  

VS2017 V15.9.5, Visual Micro V1.1811.24

TIA,

Frank

jrowberg, 2019-01-17:
I haven't seen this one before. Is Visual Micro auto-generating any code, or otherwise mangling the original? The arrangement of the main "raw-only" implementation of MPU6050 code together with the DMP-capable add-on is far from ideal, but I don't believe there is any overlap or redefinition of variables, classes, structs, etc.

jrowberg, 2019-01-17:
I don't think `struct MPU6050` appears anywhere in the original code, so if you can find that in any intermediate output, that file would be helpful to look over.

jrowberg, 2019-01-18:
Does the .ino example open and build cleanly in the Arduino IDE, as a starting point? I have not used Visual Micro in any meaningful way, so I'm not sure what it might be doing to complicate the process. Knowing that the correct code in the correct place compiles in the original IDE will be a good starting point.

paynterf, 2019-01-18:
Jeff,

I'll try that.  It may take me a while, as I haven't ever actually used the
Arduino IDE ;-)

Frank

jrowberg, 2019-01-18:
By the way, if you're not already aware, just for fun you might be interested in looking at a fledgling clean-slate attempt at fixing all of the little (and not so little) quirks about I2Cdevlib:

https://community.perilib.io/t/perilib-i2cdevlib-reborn/15

I2Cdevlib will stick around for sure, but I'm working slowly towards a better end result in the only way I think is possible, given the history, community, and complexity now tightly woven into the original project.

paynterf, 2019-01-18:
It compiles in the Arduino IDE if I first remove SBWire.h/cpp from the project folder.  Apparently the Arduino IDE compiles everything in the sketch folder, whether or not it is referenced in the main program

paynterf, 2019-01-26:
Jeff,

Wow - your new i2cdevlib project sounds great - but HUGE! ;-).  I agree
with everything you said about the difficulties with the somewhat
monolithic nature of the current i2cdevlib, and I'm looking forward to
being able to work with smaller chunks for my Arduino-based robotics
projects.

I loaded up a smaller IMU6050 test project to see if I could narrow down
the 'violation of the one definition rule' problem.  The project uses the
SBWire implementation, but I doubt if that's the culprit.  It compiles in
both VS2017/VM and the Arduino IDE, with basically the same 'one
definition' warnings.  I have attached a file containing the verbose
compiler output from the Arduino IDE run, and a file containing the code
itself.

TIA,

Frank

-- 
G.Frank Paynter, PhD
OSU ESL Research Scientist (ret)
EM Workbench LLC
614 638-6749 (cell)

In file included from C:\Users\Frank\Documents\Arduino\Arduino_IMU6050_Test4\Arduino_IMU6050_Test4.ino:30:0:

sketch\MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGravity(int16_t*, const uint8_t*)':

sketch\MPU6050_6Axis_MotionApps20.h:686:65: warning: integer overflow in expression [-Woverflow]

         - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384);

                                                                 ^

sketch\MPU6050.h:436:7: warning: type 'struct MPU6050' violates one definition rule [-Wodr]

 class MPU6050 {

       ^

sketch\MPU6050.h:436:7: note: a different type is defined in another translation unit

 class MPU6050 {

       ^

sketch\MPU6050.h:826:22: note: the first difference of corresponding definitions is field 'dmpPacketBuffer'

             uint8_t *dmpPacketBuffer;

                      ^

sketch\MPU6050.h:1028:17: note: a field with different name is defined in another translation unit

         uint8_t devAddr;

                 ^

Sketch uses 20070 bytes (62%) of program storage space. Maximum is 32256 bytes.
Global variables use 1263 bytes (61%) of dynamic memory, leaving 785 bytes for local variables. Maximum is 2048 bytes.

paynterf, 2019-01-26:
Jeff,
I think I found the reason I was getting 'One definition' violation warnings.  Apparently this started happening when the board platform manager for the AVR boards went from 1.6.21 to 1.6.22 - see this post

https://forum.arduino.cc/index.php?topic=593211.0

Frank

jrowberg, 2019-01-26:
Hi Frank,

Maybe I'm missing it, but are you sure you attached the verbose compiler output? I don't see it in your reply from yesterday afternoon.

paynterf, 2019-01-28:
> Oops - here it is

Jeff,

I have narrowed the problem down to the point where all extraneous code has
been removed, leaving only the following:

/*
    Name:       FourWD_WallE_V5.ino
    Created: 1/27/2019 12:14:57 PM
    Author:     FRANKWIN10\Frank
*/
#include <EEPROM.h>
#include <Wire.h>
#include "MPU6050_6Axis_MotionApps20.h" //06/23/18 modified for 20Hz
interrupt freq
#include "I2Cdev.h"

void setup()
{

}

void loop()
{

}

The above program will produce the 'one definition' warning in VS2017/VM,
but compiles cleanly in the Arduino IDE 1.8.7 using either 1.6.21 or 1.6.23
board manager versions (at one time I thought this was the problem, but
apparently not).

I see in the verbose output that including i2cdev.h also causes all the
stuff in the RTIMULib-Arduino-master library to be included and compiled,
and so I suspect this is where the 'one definition' warning is coming
from.  However, this warning does not occur when compiled in the IDE, even
though I would have sworn that both VS/VM and the IDE use the same exact
compiler tools (I thought all VM did was ship the source file off to the
Arduino IDE).

In any case, I have attached my minimal source file, the verbose compiler
outputs from both the VS/VM and Arduino IDE runs, and a zip file containing
my entire project folder in case there is some issue with the actual source
files included with the project.

Please let me know if this helps track this problem down - it's driving me
a little batty at the moment ;-).

TIA,

Frank

-- 
G.Frank Paynter, PhD
OSU ESL Research Scientist (ret)
EM Workbench LLC
614 638-6749 (cell)

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Frank\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Frank\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Frank\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10807 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_365529 -warnings=none -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_325438 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Frank\Documents\Arduino\FourWD_WallE_V5\FourWD_WallE_V5.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Frank\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Frank\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Frank\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10807 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_365529 -warnings=none -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_325438 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Frank\Documents\Arduino\FourWD_WallE_V5\FourWD_WallE_V5.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_365529\\sketch\\FourWD_WallE_V5.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduin [... cut ...] =16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c" -o "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Debug\core\wiring.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fda

jrowberg, 2019-01-28:
Out of curiosity, what happens if you temporarily move the RTIMU library (in whatever form(s) you have it) entirely out of the library/include path used by the IDE, so that it can't be compiled in? Is it possible that there is some MPU6050 name overlap from something in that code, and the IDE is automatically pulling it in because of some auto-include feature. Or something. I'm grasping at straws here, but it's an idea.

paynterf, 2019-01-30:
> Jeff,
>
> I've attached a file containing the entire verbose debug compile outputs
> for both DEBUG and RELEASE modes in the VS/VM environment, but essentially
> they both croak because it can't find the RTIMULib_Arduino_Master folder.
> Same thing happens in the Arduino IDE compile (no surprise)
>
> I can't see how/why/where,  but apparently the MP6050 code requires the
> RTIMULib.  Ideas?
>
> Frank

Jeff,

So this time I went back to the web page for the part I am using (DFRobots
MPU6050) and looked at their example code.  I constructed the simplest
possible program using their code, as shown below:

    Author:     FRANKWIN10\Frank
*/
#include <Wire.h>
#include "I2Cdev.h"
#include "MPU6050.h" //06/23/18 modified for 20Hz interrupt freq

MPU6050 accelgyro;

void setup()
{

}

void loop()
{

}

And compiled in RELEASE mode with Visual Micro's 'Deep Search for
Libraries' option enabled.  Voila! It compiled without any errors and
without the 'one definition rule' warning!  (verbose log attached).  Then I
went back and compiled it again with the 'Deep Search' option turned OFF,
and this time it croaked because it couldn't find EEPROM.h.

So, I added EEPROM.H to the #include list and tried again, with 'Deep
Search' turned off

#include <Wire.h>
#include <EEPROM.H>
#include "I2Cdev.h"
#include "MPU6050.h" //06/23/18 modified for 20Hz interrupt freq

MPU6050 accelgyro;

This time it compiled OK, but with a warning:

 "FourWD_WallE_V5.ino:7: In file included from
EEPROM.H: 145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
   static EEPROMClass EEPROM"

The above warning occurs whether or not 'Deep Search' is enabled.

So, I went back and removed the '#include <EEPROM.H>' and enabled 'Deep
Search', and verified that I was back to my baseline.  Yep - compiled OK
w/o any warnings.  Then I replaced #include MPU6050.H with #include
"MPU6050_6Axis_MotionApps20.h" and 'Deep Search' enabled.

#include <Wire.h>
//#include <EEPROM.H>
#include "I2Cdev.h"
//#include "MPU6050.h" //06/23/18 modified for 20Hz interrupt freq
#include "MPU6050_6Axis_MotionApps20.h"

MPU6050 accelgyro;

Now it compiles, but I'm back to the 'one definition rule' warning.

So, it appears the warning is generated by something that happens when
MotionApps20 is included, but not when just MPU6050 is included - does that
help?

So, what's the advantage of using MotionApps20 and MPU6050 over just using
MPU6050?  If it's OK to just use MPU6050 for dealing with the DFRobots
device and other I2C devices (I have 3 other I2C devices on the bus), then
[... 102 lines cut ...]
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\new.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\PluggableUSB.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\Print.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\Stream.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\Tone.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\USBCore.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\WMath.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core.a" "C:\Users\Frank\AppData\Local\Temp\VMBuilds\FourWD_WallE_V5\mega_atmega2560\Release\core\WString.cpp.o"
 
Building libraries ...

Using library Wire version 1.0 in folder "file:///C:/Program%20Files%20(x86)/Arduino/hardware/arduino/avr/libraries/Wire"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560  -Wno-unknown-pragmas -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" -I"C:\Users\Frank\Documents\Arduino\libraries\RTIMULib-Arduino-master" -I"C:\Users\Frank\Documents\Arduino\libraries\MPU6050" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" -I"C:\Program Files (x86)\Arduino\libraries" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries" -I"C:\Users\Frank\Documents\Arduino\libraries" "C:\P [... cut ...] "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" -I"C:\Users\Frank\Documents\Arduino\libraries\RTIMULib-Arduino-master" -I"C:\Users\Frank\Documents\Arduino\libraries\MPU6050" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src" -I"C:\Program Files (x86)\Arduino\libraries" -I"c:\program files (x86)\microsoft visual studio\2017\community\common7\ide\extensions\sivbitce.zrh\Micro Platforms\default\debuggers" -I"C:\Program Files (x86)\Arduino\hardware\

paynterf, 2019-02-01:
Jeff,

I've been pounding on this problem for a while now, and although I'm not
getting any closer to a solution, I have simplified the problem somewhat.
I can now reliably re-create the 'one definition rule' and overflow
warnings using just your MPU6050_DMP6 example code and Arduino IDE 1.8.7.
I have also verified that I can actually run a DFRobots MPU6050 module
using the example code, compiled using the Arduino IDE and uploaded to a
Mega 2560

So, it appears the warnings are real.  Fixing the 'overflow' error is
straightforward: Simply change '        - (int32_t)qI[2] * qI[2] +
(int32_t)qI[3] * qI[3]) / (2 * 16384);' to '        - (int32_t)qI[2] *
qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384UL);'

However, I'm still completely mystified by the 'one definition rule'
warning.  I made sure all the i2cdev files were in a completely separate
folder in a different user space even, so there could be no possibility of
some sort of library search finding the RTIMU stuff that I thought was the
cause of the is warning before, and I placed all the required .cpp/.h files
right in the sketch folder, so everything is local.  So, whatever is
causing the 'one definition' warning is in one of these files (attached).

Hope this helps,

Frank

jrowberg, 2019-02-01:
Hi Frank,

It seems the attached files are still getting stripped out of your email replies to Github issues (or at least this one). Can you post them in a comment via the web interface?

paynterf, 2019-02-01:
Jeff,

Here they are.  They might be getting stripped out by my email provider (Google Gmail), which I now recall doesn't like ZIP files :-(

[MPU6050.zip](https://github.com/jrowberg/i2cdevlib/files/2822161/MPU6050.zip)

paynterf, 2019-02-04:
Jeff,

OK, I finally have it all figured out - I think.  The 'one definition rule' (ODR) warning is caused when the compiler/linker sees code that can produce two different definitions for the same object.  If that can happen, EVER, then an ODR violation warning is issued. Believe it or not, that is exactly what happens when the compiler processes MPU6050.H - it sees that there are some conditions for which two different descriptions of the MPU6050 class could exist - and says "no no". The relevant portion of the class definition is shown below:

```
  private:
        uint8_t devAddr;
        uint8_t buffer[14];
    #if defined(MPU6050_INCLUDE_DMP_MOTIONAPPS20) or defined(MPU6050_INCLUDE_DMP_MOTIONAPPS41)
        uint8_t *dmpPacketBuffer;
        uint16_t dmpPacketSize;
    #endif

```

When the compiler sees these lines, it says to itself; "Hmm, the way this is written, it is theoretically possible for there to exist **two different versions** of 'Class MPU6050' one with just two private member variables (devAddr & buffer) and one with four (with the addition of dmpPacketBuffer & dmpPacketSize), and this is a strict no-no; I'm going to whack that programmer across the head with an ODR violation!"

If the #ifdefined and #endif lines are commented out - the ODR warning goes away

Now, I suspect nobody has ever had a problem with this issue, as it would be very unlikely to have a project where BOTH versions of MPU6050 are in play, but of course the compiler doesn't see it this way.  To address this problem, I think you might want to create a derived class (maybe MPU6050_DMP?) that simply adds the two additional private variables, or maybe better yet - simply remove the #ifdefined/#endif and let the two additional private variables tag along in every MPU6050 object, whether or not it is needed.  If not, users will continue to see the ODR warning.

On a slightly different, but related subject, the OTHER warning was due to a potential integer overrun in the dmpGetGravity() function, as shown below:

```
uint8_t MPU6050::dmpGetGravity(int16_t *data, const uint8_t* packet) {
    /* +1g corresponds to +8192, sensitivity is 2g. */
    int16_t qI[4];
    uint8_t status = dmpGetQuaternion(qI, packet);
    data[0] = ((int32_t)qI[1] * qI[3] - (int32_t)qI[0] * qI[2]) / 16384;
    data[1] = ((int32_t)qI[0] * qI[1] + (int32_t)qI[2] * qI[3]) / 16384;
    data[2] = ((int32_t)qI[0] * qI[0] - (int32_t)qI[1] * qI[1]
	       - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384);
    return status;
}

```
if the last line of the above calculation is changed to (note the addition of 'UL')
	       - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384UL);
then this warning goes away as well.

Regards,

Frank

Massyp, 2019-06-12:
Hello every body, 

I m trying to do what you did to eliminate the violations warning, but I don't don't understand how you did it. Because, I read the two library programs, and I saw that if we comment the two line as you said, the other program will not be able to use the methods defined under this program. And it display me lot of errors, 
So, if you can share us the final solution it will be helpful 

Thank you at advance

paynterf, 2019-06-12:
Massyp,

Hmm, not sure what the problem is:  If you change

 private:
        uint8_t devAddr;
        uint8_t buffer[14];
    #if defined(MPU6050_INCLUDE_DMP_MOTIONAPPS20) or defined(MPU6050_INCLUDE_DMP_MOTIONAPPS41)
        uint8_t *dmpPacketBuffer;
        uint16_t dmpPacketSize;
    #endif

to 
  private:
        uint8_t devAddr;
        uint8_t buffer[14];
        uint8_t *dmpPacketBuffer;
        uint16_t dmpPacketSize;

Eliminating the #if defined... and #endif lines, you should be good to go.

Does this help?

Frank

Massyp, 2019-06-13:
Hi Frank, 
Are you talking about the code in this link : 

https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050.h

If not, I think that I'm using the wrong one !!

Thank you for responding

Massyp, 2019-06-13:
Hi frank, 

I think that I used and version of this code : in the MPU6050.h code, they declare  " uint8_t *dmpPacketBuffer;
uint16_t dmpPacketSize;"  as a public variables inside the special methods for MotionApps 2.0 implementation. 

Now I resolve it inspiring from your advice just by cutting these to lines and declare it as a private variables as you said 

Thank you Frank

paynterf, 2019-06-13:
Massyp,

I am just repaying a VERY small part of the debt I owe to all the others who have helped me over the years.  You can thank me by helping someone else sometime in the future ;-).

Frank

closed 2019-06-13 by paynterf

Gefte, 2020-02-21:
C:\Users\Gefte\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:617:55: error: no 'uint8_t MPU6050::dmpGetCurrentFIFOPacket(uint8_t*)' member function declared in class 'MPU6050'

 uint8_t MPU6050::dmpGetCurrentFIFOPacket(uint8_t *data) { // overflow proof

                                                       ^

MathewAloisio, 2021-03-28:
Sorry to re-open an old issue but I ran into this issue and wanted to correct some of the incorrect information here. This issue is due to improper coding (not a criticsm, the libraries are amazing!). The functions in the header files that are defined and implemented directly in the header files need to be prefixed with 'inline' otherwise technically speaking it's correct for compilers to run into this issue. Some compilers may not, but it's improper to not have the functions declared as inline that are coded as such.