jrowberg/i2cdevlib #425: Shift overflow warnings?

paynterf, 2019-01-17:
Whenever I do a 'clean' build of my project  using your fine i2cdevlib, I get a large number of warnings having to do with shift count overflows, as shown below:

Compiling debug version of 'FourWD_WallE2_V1' for 'Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)'
 
`FourWD_WallE2_V1.ino:17: In file included from
MPU6050_6Axis_MotionApps20.h: In member function uint8_t MPU6050::dmpGetAccel(int32_t*, const uint8_t*)
MPU6050_6Axis_MotionApps20.h: 546:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[28] << 24) + (packet[29] << 16) + (packet[30] << 8) + packet[31])
MPU6050_6Axis_MotionApps20.h: 546:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[28] << 24) + (packet[29] << 16) + (packet[30] << 8) + packet[31])
MPU6050_6Axis_MotionApps20.h: 547:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[32] << 24) + (packet[33] << 16) + (packet[34] << 8) + packet[35])
MPU6050_6Axis_MotionApps20.h: 547:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[32] << 24) + (packet[33] << 16) + (packet[34] << 8) + packet[35])
MPU6050_6Axis_MotionApps20.h: 548:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[36] << 24) + (packet[37] << 16) + (packet[38] << 8) + packet[39])
MPU6050_6Axis_MotionApps20.h: 548:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[36] << 24) + (packet[37] << 16) + (packet[38] << 8) + packet[39])
MPU6050_6Axis_MotionApps20.h: In member function uint8_t MPU6050::dmpGetQuaternion(int32_t*, const uint8_t*)
MPU6050_6Axis_MotionApps20.h: 570:30: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[0] << 24) + (packet[1] << 16) + (packet[2] << 8) + packet[3])
MPU6050_6Axis_MotionApps20.h: 570:50: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[0] << 24) + (packet[1] << 16) + (packet[2] << 8) + packet[3])
MPU6050_6Axis_MotionApps20.h: 571:30: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[4] << 24) + (packet[5] << 16) + (packet[6] << 8) + packet[7])
MPU6050_6Axis_MotionApps20.h: 571:50: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[4] << 24) + (packet[5] << 16) + (packet[6] << 8) + packet[7])
MPU6050_6Axis_MotionApps20.h: 572:30: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[8] << 24) + (packet[9] << 16) + (packet[10] << 8) + packet[11])
MPU6050_6Axis_MotionApps20.h: 572:50: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[8] << 24) + (packet[9] << 16) + (packet[10] << 8) + packet[11])
MPU6050_6Axis_MotionApps20.h: 573:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[3] = ((packet[12] << 24) + (packet[13] << 16) + (packet[14] << 8) + packet[15])
MPU6050_6Axis_MotionApps20.h: 573:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[3] = ((packet[12] << 24) + (packet[13] << 16) + (packet[14] << 8) + packet[15])
MPU6050_6Axis_MotionApps20.h: In member function uint8_t MPU6050::dmpGetGyro(int32_t*, const uint8_t*)
MPU6050_6Axis_MotionApps20.h: 603:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[16] << 24) + (packet[17] << 16) + (packet[18] << 8) + packet[19])
MPU6050_6Axis_MotionApps20.h: 603:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[0] = ((packet[16] << 24) + (packet[17] << 16) + (packet[18] << 8) + packet[19])
MPU6050_6Axis_MotionApps20.h: 604:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[20] << 24) + (packet[21] << 16) + (packet[22] << 8) + packet[23])
MPU6050_6Axis_MotionApps20.h: 604:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[1] = ((packet[20] << 24) + (packet[21] << 16) + (packet[22] << 8) + packet[23])
MPU6050_6Axis_MotionApps20.h: 605:31: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[24] << 24) + (packet[25] << 16) + (packet[26] << 8) + packet[27])
MPU6050_6Axis_MotionApps20.h: 605:52: warning: left shift count >= width of type [-Wshift-count-overflow]
   data[2] = ((packet[24] << 24) + (packet[25] << 16) + (packet[26] << 8) + packet[27])
`

I know I can suppress these warnings by including '-Wshift-count-overflow' in my compiler flag list, but I'm not sure I should.  Are these just 'nuisance' warnings, or an indication of something serious?

VS2017 V15.9.5, Visual Micro V1.1811.24

TIA,

Frank

unresfriado, 2019-04-14:
i get exactly the same

paynterf, 2019-04-14:
unresfriado,

See https://github.com/jrowberg/i2cdevlib/issues/424 for more than you ever wanted to know about the ODR violation ;-).

Frank

unresfriado, 2019-04-21:
> unresfriado,
> 
> See #424 for more than you ever wanted to know about the ODR violation ;-).
> 
> Frank

Frank,

I've tried it and now i get this error:

`In file included from C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:42:0,

                 from C:\Users\Vlad\Desktop\xd\arduinos\aBalncebot\Balancebot\Balancebot.ino:4:

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050.h:1025:37: error: expected unqualified-id before ')' token

    (MPU6050_INCLUDE_DMP_MOTIONAPPS20) or defined(MPU6050_INCLUDE_DMP_MOTIONAPPS41)

                                     ^

In file included from C:\Users\Vlad\Desktop\xd\arduinos\aBalncebot\Balancebot\Balancebot.ino:4:0:

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetAccel(int32_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:581:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetAccel(int16_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:589:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetAccel(VectorInt16*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:597:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetQuaternion(int32_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:605:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetQuaternion(int16_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:614:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGyro(int32_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:638:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGyro(int16_t*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:646:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h: In member function 'uint8_t MPU6050::dmpGetGyro(VectorInt16*, const uint8_t*)':

C:\Users\Vlad\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:654:31: error: 'dmpPacketBuffer' was not declared in this scope

     if (packet == 0) packet = dmpPacketBuffer;

                               ^

exit status 1`

jrowberg, 2021-09-28:
All ODR/overflow warnings should be fixed with the most recent commit.

closed 2021-09-28 by jrowberg