jrowberg/i2cdevlib #430: physical interpretation of the quaternion from MPU6050_DMP6.ino code

shomedas, 2019-02-26:
Unit quaternions denote 3D rotation and can be converted to a 3D rotation matrix. Does the unit quaternion from the MPU6050 DMP code denote a rotation from an arbitrary cordinate frame that is fixed for every initialization of the IMU ? 
The relative quaternion between two quaternions gives the rotation between the two quaternions
If two quaternion readings are taken from two different orientations of the IMU we can get a relative quaternion or a relative rotation matrix. What can be said about the relative quaternion(or rotation matrix) computed to the actual rotation of the IMU from its initial orientation?

jrowberg, 2019-02-26:
After boot-up calibration, the DMP provides an absolute position based on pitch and roll (the values affected by gravity), but the yaw/heading value is an arbitrary point where "zero" is the initial rotation when the DMP starts running, right after the binary load. Since it has no magnetometer, it has nothing to use for absolute heading reference.

shomedas, 2019-02-26:
If two quaternion readings are taken from two different orientations of the IMU what does the relative quaternion corresponds to?

My attempt to find the relative rotation between two orientations is as below

I place my IMU on a table with the IMU facing up. Then I rotate it along the y axis(along the y line marked on the IMU breakout board)  by 90 degrees. The quaternions captured at the initial and the final orientation are q0 & q1. the relative quaternion qr is = q1*q0_inverse.

Quaternionf q0(0.83,0.01,-0.03,0.55);
Quaternionf q1(0.60,-0.38,0.59,0.39);

relative quaternion, qr=0.692051,-0.6586,0.295248,-0.011818

The rotation matrix and the axis angle is created from qr using the eigen library.

Since the rotation is 90 about y axis i expect a rotation matrix(from the relative quaternion) of the form
[cos(theta) 0     sin(theta)                 [0 0 1
        0       1          0                OR    0 1 0
-sin(theta)  0    cos(theta)]                -1 0 0]

But I get

M=  
0.825377  -0.372544   0.424221
-0.405259   0.132212   0.904591
-0.393087  -0.918548 -0.0418515

in axis angle format it should result to angle of 90 deg and axis = (0 1 0)

However I get
angle= 84.3567

axis =
-0.91238
0.409017
-0.0163718

the axis found is predominantly oriented along x axis unlike how i rotated it ie along y axis.

The IMU was well calibrated and offsets applied to the code as expected

i3130002, 2019-08-13:
@jrowberg 
> After boot-up calibration, the DMP provides an absolute position based on pitch and roll (the values affected by gravity), but the yaw/heading value is an arbitrary point where "zero" is the initial [...]

I was impressed with this behavior until I ran a simulator off of it! To me it seems both ways! The Z-axis of rotation is exactly as you mentioned though X & Y axis goes back to zero as soon as rotation stops! 
(I'm having the board MPC6050 perpendicular to  bread board which according to it's drawings X is up (Similar to this setup in https://forum.arduino.cc/index.php?action=dlattach;topic=271519.0;attach=99086)

[image]