jrowberg/i2cdevlib #431: MPU6050 on Arduino MEGA Verify failed

Alieno79, 2019-02-27:
Ciao a tutti, non riesco a capire cosa non vada
Ho scaricato il pacchetto JROWBERG-i2cdevlib-master.zip
Messo nella cartella delle librerie i2cdev e mpu6050
Quando faccio la verifica mi indica il seguente errore :

`D:\Documenti\Arduino\libraries\MPU6050/MPU6050.h:436:7: warning: type 'struct MPU6050' violates one definition rule [-Wodr]

 class MPU6050 {

       ^

D:\Documenti\Arduino\libraries\MPU6050\MPU6050.h:436:7: note: a different type is defined in another translation unit

 class MPU6050 {

       ^

D:\Documenti\Arduino\libraries\MPU6050/MPU6050.h:1026:18: note: the first difference of corresponding definitions is field 'dmpPacketBuffer'

         uint8_t *dmpPacketBuffer;

                  ^

D:\Documenti\Arduino\libraries\MPU6050\MPU6050.h:436:7: note: a type with different number of fields is defined in another translation unit

 class MPU6050 {
`

Ho cercato per i forum, e parte del problema si era risolto modificando una stringa cosi : 2 * 16384L

Ma resta l'errore sopra indicato.. potete per favore aiutarmi a capire?

Grazie anticipatamente a tutti!!

jrowberg, 2019-02-27:
Duplicate of #424

closed 2019-02-27 by jrowberg

Alieno79, 2019-02-27:
Sorry, but i don't understand the solution on post #424 ..

jrowberg, 2019-02-27:
In short, the one-definition rule warning can be ignored, and is due to the non-optimal arrangement between multiple headers and the source file for MPU6050 support. Eliminating the warning would require refactoring the library so that either the DMP-enabled implementation is a subclass (ideal), or else the DMP code is always available rather than selectively available by different `#include` directives, which would (as written) result in wasted flash usage for people who don't need the DMP.

The overflow warning is solved by typecasting expressions to `(int32_t)` so that the mathematical operations cannot possibly result in an intermediate value larger than the 16-bit containers.

Alieno79, 2019-02-27:
Thank you, but program is freeze on console message : 
`Initializing DMP...
Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
`
Raw data with other examples it's ok

you have any suggestions?

Sorry for my english.. i speak italian ahaha

jrowberg, 2019-02-27:
It looks like the interrupt pin/wire may not be connected properly. The raw example uses polling, so no interrupts are required. Have you verified that the IMU's INT pin is connected to the correct Arduino pin?

Alieno79, 2019-02-27:
I changed the cable that connected pin 2 .. now it works !!!
Thank you for your precious help !!!
Good job!! :)