jrowberg/i2cdevlib #434: Error Compiling for Board Arduino Nano

Curious-Nikhil, 2019-03-20:
I am running the **MPU6050_DMP** Example for **Arduino Nano.** 
**Processor** - ATmega328P (Old Bootloader)

The lib are all up to date.

This error message doesn't make any sense at all.

Hopefully there's a fix to this. 
I'll try some troublshooting and update this post. 

Thanks
[image]

```
Arduino: 1.8.5 (Windows 10), Board: "Arduino Nano, ATmega328P (Old Bootloader)"

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::I2Cdev()'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::I2Cdev()'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeBytes(unsigned char, unsigned char, unsigned char, unsigned char*)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeByte(unsigned char, unsigned char, unsigned char)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeWords(unsigned char, unsigned char, unsigned char, unsigned int*)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeWord(unsigned char, unsigned char, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readBytes(unsigned char, unsigned char, unsigned char, unsigned char*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readByte(unsigned char, unsigned char, unsigned char*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readBit(unsigned char, unsigned char, unsigned char, unsigned char*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readBits(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeBit(unsigned char, unsigned char, unsigned char, unsigned char)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readTimeout'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeBits(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readWords(unsigned char, unsigned char, unsigned char, unsigned int*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readWord(unsigned char, unsigned char, unsigned int*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readBitW(unsigned char, unsigned char, unsigned char, unsigned int*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::readBitsW(unsigned char, unsigned char, unsigned char, unsigned char, unsigned int*, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeBitW(unsigned char, unsigned char, unsigned char, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\MPU6050\I2Cdev.cpp.o (symbol from plugin): In function `I2Cdev::I2Cdev()':

(.text+0x0): multiple definition of `I2Cdev::writeBitsW(unsigned char, unsigned char, unsigned char, unsigned char, unsigned int)'

libraries\I2Cdev\I2Cdev.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "I2Cdev.h"
 Used: C:\Program Files (x86)\Arduino\libraries\I2Cdev
 Not used: C:\Users\mishr\Documents\Arduino\libraries\MPU6050
exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

per1234, 2019-03-21:
We will likely resolve this in Curious-Nikhil's post on the Arduino Forum:
http://forum.arduino.cc/index.php?topic=604834

Curious-Nikhil, 2019-03-25:
**Fix** - It was fixed by a simple reinstall of Arduino and libraries. 

**Reason** - No Idea.

closed 2019-03-25 by Curious-Nikhil