jrowberg/i2cdevlib #437: Calibration with sensor mounted upside down not correct

hspeedy, 2019-04-27:
I have to mount the sensor upside down, so orientation of z-axis is inverted.
Calibration with normal sensor orientation works well. But when mounted upside
down, i get wrong g values for z-axis. 1g and -3g after rotating the sensor to normal
orientation.
Does anybody know how to change the code to proper with an upside down mounted
sensor? Please not, I can't calibrate the sensor in normal orientation to get the values.
I might need to recalibrate later and can't change orientation since everything is mounted
and would take a lot of time to unmount.

Thanks for any replies.

ZHomeSlice, 2019-07-11:
In order to calibrate the MPU6050 Gravity is mathematically removed assuming the MPU is in the correct orientation. this allows us to get all zeros (or as close to as possible) with the accelerometer.  So to calibrate the MPU6050, place it in its normal right side up orientation then use these offsets to replace the defaults and then flip the MPU6050 upside down. If you are required to calibrate the MPU6050 in any other orientation you will need to change the program to correctly remove gravity when performing the offset math.

yo90bosses, 2019-10-08:
@ZHomeSlice im having a similar problem to the original poster. Im using the DMP and the sensor is mounted upside down. The problem is though that the pitch and roll is then at startup at zero (it should be 180 or so) and the data it totally unreliable. Is this normal?