jrowberg/i2cdevlib #438: specified folder/zip file does not contain a valid library

germain171, 2019-05-01:
I keep getting this issue with i2cdevlib-master.zip. when i try to add zip to library via Add zip feature in Arduino IDE.

I think i have a fresh download from git. the error i am getting that is needing me to try compile the code in the IDE is:
MPU6050_DMP6_ESPWiFi_dave_2:47:20: error: I2Cdev.h: No such file or directory

 #include <I2Cdev.h>

                    ^

compilation terminated.

exit status 1
I2Cdev.h: No such file or directory

So i try install the Library and get 'Specified folder/zip file does not contain a valid library'

to try resolve this i have removed everything i can find regarding Libraries and Arduino, all appf folders and clean reg ect.
downloaded the latest hourly build 91.8.10 2019/04/18 12:33 - still cannot install.

now the real kicker for me is that at home on my main PC i have it all running perfectly. Complies and uploads working sketch. This error is only on my work laptop - using at the weekend when going traveling.

samanshams2009, 2020-10-21:
I am getting same isse
Specified folder/zip file does not contain a valid library

Kwieeciol, 2021-08-10:
Me too, any fixes?