jrowberg/i2cdevlib #441: how to use MPU6050 dmp6 without IRQ pin?

dsyleixa, 2019-05-09:
hello,
how to use MPU6050 dmp6 without an IRQ pin?
I don't have free GPIOs on my M4 board any more (Adafruit Feather).
Instead of IRQ, I want to repetitively read i2c data in the loop() either if new data are already available or not.

ZHomeSlice, 2019-07-25:
The DMP calculates values every 10 milliseconds just set a timer and check for data. if there is no data wait a millisecond. and check again this works well and is a great alternative to using the IRQ. 
I Like using a blink without delay timer like this one:
`  static unsigned long _ExactTimer; 
// This timer is accurate and is what I would use
// warning this one can wind up and repeat itself over and over until it catches up
  if ( millis() - _ExactTimer>= (10)) {
    _ExactTimer+= (10);
    // Get DMP
  }

  static unsigned long _AfterTimer;
// This one should work but you will not be as accurate as another code execution time is added to the timer 10ms + code execution time
// warning this one is not accurate and will start the timer after everything else 
  if ((millis() - _AfterTimer) >= (10)) {
    _AfterTimer= millis();

  }`

dsyleixa, 2019-07-27:
I have my own repetitive loops to call different devices (by i2c, 1-wire, ADC, UART, and SPI), and the MPU is one of them.  1 loops lasts about overall 15ms.
As stated, I have no GPIO pin for IRQ left, and I know that there must be ways to drive the MPU WITHOUT using IRQ pins.

So how to do that here without IRQ pin  using this dmp6 lib?

paynterf, 2019-10-08:
I have a non-interrupt example at https://www.fpaynter.com/2019/10/polling-vs-interrupt-with-mpu6050-gy-521-and-arduino/

dsyleixa, 2019-10-09:
> functions
> 
> I have a non-interrupt example at https://www.fpaynter.com/2019/10/polling-vs-interrupt-with-mpu6050-gy-521-and-arduino/

hello,
thank you for your link!
Does your code use the internal dmp processing function or does it process raw data?

dsyleixa, 2019-10-09:
thank you, I just saw that you are also using the dmp functions! I'll have to re-assemble and re-wire everything anew and then try it out ASAP!

dsyleixa, 2019-10-09:
thank you, I just saw that you are also using the dmp functions! I'll have to re-assemble and re-wire everything anew and then try it out ASAP!
where can your libs and files  be downloaded, preferably on github?

paynterf, 2019-10-09:
dyslexia,

Glad you found it useful.  No libraries, nothing on Github, but the
complete Arduino program is there in the post ;-).

Frank

ZHomeSlice, 2019-10-09:
Excellent Example @paynterf ,
I noticed that you are continually polling The mpu6050 over the i2c bus for the dmpPacketAvailable if you would like you could just wait till about 9 ms passes and then start polling. The DMP triggers every 10 ms like clockwork

example:

```
    static unsigned long BlinkTimer;
    if ((millis() - BlinkTimer) >= (9)) { // After FIFO read no new data will be there for at least 9 ms
	if (mpu.dmpPacketAvailable())
	{
		global_yawval = GetIMUHeadingDeg(); //retreive the most current yaw value from IMU

		// blink LED to indicate activity
		blinkState = !blinkState;
		digitalWrite(LED_PIN, blinkState);
                BlinkTimer= millis();  // Reset the timer for another 9ms delay
	}
    }
```

paynterf, 2019-10-09:
Thanks for the compliment - but it's a pretty small contribution compared
to what you have done! ;-)

I set the loop up this way because all the 'work' occurs in the 'other
program stuff' block, which executes every IMU_CHECK_INTERVAL_MSEC
milliseconds.  The next check for dmp availability may not occur until
after several 10's of milliseconds have passed so I didn't think there was
any good reason to not check as soon as possible and as often as possible.
After all, when the program is 'idling' in the program loop between 'other
program stuff' passes, there is nothing else going on anyway.

Regards,

Frank

dsyleixa, 2019-10-10:
sorry, perhaps Im too stupid...
IIUC, in https://www.fpaynter.com/2019/10/polling-vs-interrupt-with-mpu6050-gy-521-and-arduino/ you are using some libs
```
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps_V6_12.h"  //changed to this version 10/04/19
#include "elapsedMillis.h"
```
where can they be found?

dsyleixa, 2019-10-10:
sorry, can't find it! :(
do you have a link?

dsyleixa, 2019-10-10:
of course I now that repo, but where is the file
elapsedMillis.h
??

paynterf, 2019-10-10:
That library is part of the basic Arduino package

dsyleixa, 2019-10-11:
I know of course, my q
uestion was:
whre is the file 
elapsedMillis.h
?

MPU6050 and i2cdev libs are already installed.The Arduino IDE can't find  elapsedMillis.h

please link to that specific file!

paynterf, 2019-10-11:
dyslexia,

a Google search of 'Arduino ElapsedMillis' resulted in 18,400 hits in 0.52
seconds. It would have been much faster for both of us if you had simply
done that. I'm not sure how you expect to be a successful Arduino
programmer if you expect others to do all your work for you.

Frank

dsyleixa, 2019-10-11:
sorry, but you falsely wrote that it should be another lib by JRowberg, not a 3rd party lib, so of course I only searched in THIS repo.

dsyleixa, 2019-10-11:
nonetheless, now having successfully #included the elapsedMillis, lib, the code gives a compile error:

> exit status 1
> '_BV' was not declared in this scope

line 161
`  if ((mpuIntStatus & _BV(MPU6050_INTERRUPT_FIFO_OFLOW_BIT)) || fifoCount >= 1024)`

dsyleixa, 2019-10-11:
2nd,
if trying that there are additional issues about dtostrf and  printfloatx("", Data[2], 5, 0, "\n");

dsyleixa, 2019-10-11:
Update:

I added
#include <avr/dtostrf.h>
in MPU6050.cpp

now the code can be compiled but immediately after Serial.print:
Initializing MPU6050...

it hangs up completely and freezes...

Using MPU6050_dmp6 (original example using an IRQ) is functioning fine though.

dsyleixa, 2019-10-12:
current test code (still hangs up and freezes):

```
/*
 * Name:       Arduino MPU6050 Polling Test.ino
    Created:  10/5/2019 1:31:08 PM
    Author:     FRANKWIN10\Frank
 
  This is a complete, working MPU6050 (GY-521) example using polling vs interrupts.
  It is based on Jeff Rowberg's MPU6050_DMP6_using_DMP_V6.12 example.  After confirming
  that the example worked properly using interrupts, I modified it to remove the need
  for the interrupt line.  
*/
 
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps_V6_12.h"  //changed to this version 10/04/19
#include "elapsedMillis.h"
#include <avr/dtostrf.h>

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

#define _BV(bit) (1 << (bit))  
#define LED_PIN  LED_BUILTIN // (Arduino is 13, Teensy is 11, Teensy++ is 6)

 
bool blinkState = false;
 
// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer
 
// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector
 
//extra stuff
int IMU_CHECK_INTERVAL_MSEC = 100;
elapsedMillis sinceLastIMUCheck;
float global_yawval = 0.0; //contains most recent yaw value from IMU
int global_fifo_count = 0; //made global so can monitor from outside GetIMUHeadingDeg() fcn
 
// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================
 
void setup()
{
  // I2C bus init done in SBWIRE.h
 
  Serial.begin(115200);
  while (!Serial); // wait for Leonardo enumeration, others continue immediately
 
  // initialize device
  Serial.println(F("Initializing MPU6050..."));
  mpu.initialize();
 
[... 101 lines cut ...]
    // otherwise, check for DMP data ready interrupt (this should happen frequently)
  }
  else if (mpuIntStatus & _BV(MPU6050_INTERRUPT_DMP_INT_BIT))
  {
    // read all available packets from FIFO
    while (fifoCount >= packetSize) // Lets catch up to NOW, in case someone is using the dreaded delay()!
    {
      mpu.getFIFOBytes(fifoBuffer, packetSize);
      // track FIFO count here in case there is > 1 packet available
      // (this lets us immediately read more without waiting for an interrupt)
      fifoCount -= packetSize;
    }
    global_fifo_count = mpu.getFIFOCount(); //should be zero here
 
    // display Euler angles in degrees
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    mpu.dmpGetGravity(&gravity, &q);
    mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
  }
 
  float yawval = ypr[0] * 180 / M_PI;
  return  yawval;
}

```

dsyleixa, 2019-10-29:
any ideas why the proposed code hangs up?

ZHomeSlice, 2019-10-29:
> any ideas why the proposed code hangs up?

What can you tell me about your setup?
I have tested this on MPU6050 and an MPU9250 AKA MPU6500 with a magnetometer.

Z

dsyleixa, 2019-10-30:
I have a mpu6050 breakout board from China and a Adafruit Feather M4 (!)
I'll test your code ASAP!

dsyleixa, 2019-10-30:
I have a mpu6050 breakout board from China and a Adafruit Feather M4 (!)
which is the correct setup() code?

ZHomeSlice, 2019-10-30:
top examples need to be inserted into the mpu6050 library
the bottom two examples are for your sketch and are not part of the library
note how the "mpu." object pointer is missing in the other example in the function it isn't needed in the first because it is part of the class. In the second function, it is needed because it is part of your sketch. same with the bottom 2 loop functions. 
the default setup that comes with the code works I am assuming you haven't changed that.

Z

ZHomeSlice, 2019-10-30:
Use this library with this modified example code:

```
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v6.12)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
//
// Changelog:
//      2019-07-10 - Uses the new version of the DMP Firmware V6.12
//                 - Note: I believe the Teapot demo is broken with this versin as
//                 - the fifo buffer structure has changed
//      2016-04-18 - Eliminated a potential infinite loop
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
  I2Cdev device library code is placed under the MIT license
  Copyright (c) 2012 Jeff Rowberg

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  ===============================================
*/

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps_V6_12.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

[... 278 lines cut ...]
  Serial.println(aaWorld.z);
#endif

#ifdef OUTPUT_TEAPOT
  // display quaternion values in InvenSense Teapot demo format:
  teapotPacket[2] = fifoBuffer[0];
  teapotPacket[3] = fifoBuffer[1];
  teapotPacket[4] = fifoBuffer[4];
  teapotPacket[5] = fifoBuffer[5];
  teapotPacket[6] = fifoBuffer[8];
  teapotPacket[7] = fifoBuffer[9];
  teapotPacket[8] = fifoBuffer[12];
  teapotPacket[9] = fifoBuffer[13];
  Serial.write(teapotPacket, 14);
  teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
#endif

  // blink LED to indicate activity
  blinkState = !blinkState;
  digitalWrite(LED_PIN, blinkState);

}
```

Z

dsyleixa, 2019-10-31:
thank you very much!
Unfortunately it cannot be compiled - did you test it for ARM Cortex M4 (SAMD51)?
```

Arduino: 1.8.9 (Windows 7), Board: "Adafruit Feather M4 Express (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0X239A_0X8022 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_505351 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose D:\Arduino\portable\sketchbook\mpu6050_noIRQ_000\mpu6050_noIRQ_000.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0X239A_0X8022 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_505351 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose D:\Arduino\portable\sketchbook\mpu6050_noIRQ_000\mpu6050_noIRQ_000.ino
Using board 'adafruit_feather_m4' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"mpu6050_noIRQ_000.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10809 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino" "-ID:\\arduino\\portable\\packa [... cut ...]
D:\arduino\portable\sketchbook\libraries\MPU6050\MPU6050 - old.cpp:3329:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[2], 5, 0, "\n");

  ^

Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit Feather M4 Express (SAMD51).
```

dsyleixa, 2019-10-31:
update:
thank you, I had to add 
#include <avr/dtostrf.h>
both in ino code and in MPU6050.cpp, and now it produces apparently trustworthy values!

again, thank you very much!

perhaps it might be a good idea to add this functionality to extra and enhanced libs and examples!

dsyleixa, 2019-10-31:
2 point:
`
 #define INTERRUPT_PIN 2  // use pin 2 on Arduino Uno & most boards`
can that be dropped?

`#define LED_PIN 13 // (Arduino is 13, Teensy is 11, Teensy++ is 6)`
better: 
#define LED_PIN  LED_BUILTIN // LED_BUILTIN is an  Arduino Macro for ALL Arduino Boards!

ZHomeSlice, 2019-10-31:
ya remove the useless code.

I would also use my calibration routine it is fast and accurate Change the (6) to a higher number for more detailed calibration say 10 to 20 would be sweet. 
set it on a flat level surface and run:
```
    mpu.CalibrateAccel(6);
    mpu.CalibrateGyro(6);
Serial.println();
    mpu.PrintActiveOffsets();
```

Then remove the calibration code and use what it gives you in the serial monitor for the following settings.
to do:
The values here are what works on my MPU they must be changed to match you specific MPU:
```
  mpu.setXGyroOffset(51);
  mpu.setYGyroOffset(8);
  mpu.setZGyroOffset(21);
  mpu.setXAccelOffset(1150);
  mpu.setYAccelOffset(-50);
  mpu.setZAccelOffset(1060);
```
z

dsyleixa, 2019-10-31:
so DIUC that there are auto_calibrate()  with auto_offset() functions available?

ZHomeSlice, 2019-11-01:
> so DIUC that there are auto_calibrate() with auto_offset() functions available?

Yes for gyro and if you have the MPU level yes for the accelerometer.

To use the calibration for accelerometer an absolute constant direction for gravity is needed so you need to have it level.
Ss for the Gyro, the DMP V6.12 has a calibration routine built-in. just keep it still for @ 8 seconds and you are calibrated you can see the readings stabilize further. By initially using my Calibration routine with the fixed offsets enhances the calibration of both the gyro and accelerometer as it retrieves the current offsets before performing any further tuning. so setting the fixed offsets to get them close is excellent. I noticed that the environment causes changes in the offsets. just a little but it is enough to cause a slow wandering over time one way or another temperature is the biggest issue I've discovered. 

What are the offsets
The offsets are an adjustment that literally adds value to the reading taken. that is why most average a bunch of readings to get them. when the ideal condition is present the raw values should all be zero. except for z acceleration which should be 1G I use a PI of the PID to land the offsets as each new reading taken has can be used to test the new offsets as the PID routine adjusts them.
mpu.CalibrateAccess(6) would loop about 600 times but its self advancing and the more loops the more detailed tuning parameters are used. I have it so that it will land about anything in 6 sets of 100 running it to 10 or even a little of overkill to 20 just proofs the offsets.

Hope this helps :)
Z

dsyleixa, 2019-11-01:
thank you, I will have to think about that a little while to understand it and then know what to do and how to use it ! 8)

dsyleixa, 2019-12-05:
hi,
I now tried this ode with my ES32 but now again many compile errors:

```
Sketch wird kompiliert...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spiffs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcp_transport" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcpip_adapter" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ulp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/vfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wear_levelling" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wifi_provisioning" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wpa_supplicant" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/xtensa-debug-module" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/ [... cut ...]
In file included from D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/WString.h:29:0,

                 from D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:146,

                 from C:\Users\hw\AppData\Local\Temp\arduino_build_957913\sketch\MPU6050_DMP6_noIRQ.ino.cpp:1:

D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'

 typedef unsigned long prog_uint32_t;

                       ^

```

dsyleixa, 2019-12-16:
any ideas how to fix that issue for  ESP32 boards?

ZHomeSlice, 2019-12-17:
prog stands for ProgMemory Try removing all prog_ so everything will load into regular memory. 

Z

dsyleixa, 2019-12-17:
ok, I see, thank you, 
as mentioned here:
https://github.com/jrowberg/i2cdevlib/issues/500#issuecomment-566435408

dsyleixa, 2020-02-05:
will there be a lib and example update about how to use MPU6050 dmp6 without IRQ pin ( ARM Cortex SAM, SAMD, ESP) ?

ZHomeSlice, 2020-02-05:
https://github.com/jrowberg/i2cdevlib/pull/509
The request is in place to update the files for what you are asking. Try it out leave some comments and Jeff may accept it sooner. He usually is faster with my updates.
Z

dsyleixa, 2020-02-06:
thank you very much for your efforts! As I just observed, Jeff already took notice of your PR!

ZHomeSlice, 2020-02-06:
This update works incredibly well with interrupts also. Because it can handle any buffer fill state it can be used in  any desired fashion. If you wish to check your Pentagon every 5 minutes call this function and it returns a good reading. I've not found s way to get it to return corrupted data yet..... Knock on wood :)
Z

dsyleixa, 2020-02-06:
which is the new example code, nor using a IRQ Pin ever?

ZHomeSlice, 2020-02-06:
You should now use:
`mpu.dmpGetCurrentFIFOPacket(data) ;`
Call this function when you need the latest packet. If no data is available it returns zero
So you could:
````
if(mpu.dmpGetCurrentFIFOPacket(data)) { 
//your code here to use the data
}
````
New data is available every 10ms from the MPU. This function will retrieve the most current data. 

Z

dsyleixa, 2020-02-06:
hmmm, the current example source codes are quite complicated and overburdened with preprocessor directives and unneeded optional extra functions and tons of comments, so  I can't see through all those functionalities. 
I just needed a stripped-down simplfied pure elementary example to just retrieve yaw+pitch+roll, compatible to ESP and ARM cores. Which would be a link to a source code to start from for this purpose?

ZHomeSlice, 2020-02-07:
dsyleixa,
So let's see if Jeff will add this to his MPU6050 examples but before I do so. Would you try it out and let me know what you think. Spell checking and such along with any help in un-GEEKing the verbiage would be appreciated. The hope is that this will be the first sketch newcomers use with the idea it will be self-explanatory and not overly complicated.

Z

```
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v6.12)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
//
// Changelog:
//      2020-02-02 - DMP6 Simplified Based on MPU6050_DMP6_using_DMP_V6.12

/* ============================================
  I2Cdev device library code is placed under the MIT license
  Copyright (c) 2012 Jeff Rowberg

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  ===============================================
*/

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps_V6_12.h"

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include "Wire.h"
#endif

// *****************  Setup  *****************

// uncomment the next line if you would like to use interrupts (this is no longer required)
// #define INTERRUPT_PIN 2  // use pin 2 on Arduino Uno & most boards

// comment this line if you would like to remove pausing
#define PauseForInstructions

// comment this line to skip Calibration routines you should have your own offsets set before doing so.
#define CalibrateMPU

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

/* =========================================================================
   NOTE: Arduino v1.0.1 with the Leonardo board generates a compile error
   when using Serial.write(buf, len). The Teapot output uses this method.
[... 163 lines cut ...]
    Serial.print(ypr[2] * 180 / M_PI);
    /*  All Data Available:
      mpu.dmpGetAccel(&aa, fifoBuffer);
      Serial.print("\tRaw Accl XYZ\t");
      Serial.print(aa.x);
      Serial.print("\t");
      Serial.print(aa.y);
      Serial.print("\t");
      Serial.print(aa.z);
      mpu.dmpGetGyro(&gy, fifoBuffer);
      Serial.print("\tRaw Gyro XYZ\t");
      Serial.print(gy.x);
      Serial.print("\t");
      Serial.print(gy.y);
      Serial.print("\t");
      Serial.print(gy.z);
    */
    Serial.println();

    // blink LED to indicate activity
    blinkState = !blinkState;
    digitalWrite(LED_PIN, blinkState);
  }
}
```

dsyleixa, 2020-02-07:
compile error:

```
D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651848 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_Test\MPU6050dmp6nIRQ\MPU6050dmp6nIRQ001\MPU6050dmp6nIRQ001.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651848 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_Test\MPU6050dmp6nIRQ\MPU6050dmp6nIRQ001\MPU6050dmp6nIRQ001.ino
Using board 'featheresp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-ID:\\arduino\\portable\\packages\\esp32\\hard [... cut ...]
D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'

 typedef unsigned long prog_uint32_t;

                       ^

Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

```

ZHomeSlice, 2020-02-08:
Removing these lines from MPU6050_6Axis_MotionApps_V6_12.h  should solve the problem:

        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;

dsyleixa, 2020-02-08:
tbh, i would never dare to mess around in original library code, i would be too worried about accidentally destroying something irretrievably.
So the lib actually would be expected to work for ESP and ARM cores out of the box.

ZHomeSlice, 2020-02-08:
Of course poor or incorrect programming practices pop up all over the place. I'm not a programming master but just a hobbiest that loves robotics. Also note it's not my library and I'm not an expert at c++ to be the one to determine this or how to fix it for all processor types. I just Googled your error message and seen the possible solution. I hope it help you get it working on your ESP32.
Z

dsyleixa, 2020-02-08:
perhaps it might be possible via
#ifdef __AVR__ 
typedef int8_t prog_int8_t;
    typedef uint8_t prog_uint8_t;
    typedef int16_t prog_int16_t;
    typedef uint16_t prog_uint16_t;
    typedef int32_t prog_int32_t;
    typedef uint32_t prog_uint32_t;
#endif 
??
or drop those redefinitions entirely? Actualy, who in the world still finally needs 2kb AVRs?
Just an idea, I am a simple Arduino user, not a developer as already stated.

ZHomeSlice, 2020-02-08:
Heart point. Did the change work for your ESP32? I have no way of testing my atmega328 my system isn't affected by having this in place. 
```

#ifdef __AVR__
typedef int8_t prog_int8_t;
typedef uint8_t prog_uint8_t;
typedef int16_t prog_int16_t;
typedef uint16_t prog_uint16_t;
typedef int32_t prog_int32_t;
typedef uint32_t prog_uint32_t;
#endif
```
I'll make the suggestion of you say it worked
Z

dsyleixa, 2020-02-08:
I don't know how to do that because there are complicated nested directives
#ifdef __AVR__
    #include <avr/pgmspace.h>
#else
    // Teensy 3.0 library conditional PROGMEM code from Paul Stoffregen
    #ifndef __PGMSPACE_H_
        #define __PGMSPACE_H_ 1
        #include <inttypes.h>

        #define PROGMEM
        #define PGM_P  const char *
        #define PSTR(str) (str)
        #define F(x) x

        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
        
        #define strcpy_P(dest, src) strcpy((dest), (src))
        #define strcat_P(dest, src) strcat((dest), (src))
        #define strcmp_P(a, b) strcmp((a), (b))
        
        #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
        #define pgm_read_word(addr) (*(const unsigned short *)(addr))
        #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
        #define pgm_read_float(addr) (*(const float *)(addr))
        
        #define pgm_read_byte_near(addr) pgm_read_byte(addr)
        #define pgm_read_word_near(addr) pgm_read_word(addr)
        #define pgm_read_dword_near(addr) pgm_read_dword(addr)
        #define pgm_read_float_near(addr) pgm_read_float(addr)
        #define pgm_read_byte_far(addr) pgm_read_byte(addr)
        #define pgm_read_word_far(addr) pgm_read_word(addr)
        #define pgm_read_dword_far(addr) pgm_read_dword(addr)
        #define pgm_read_float_far(addr) pgm_read_float(addr)
    #endif
#endif

ZHomeSlice, 2020-02-08:
This should be all these is used for avr processors

```
#ifdef __AVR__
    #include <avr/pgmspace.h>
#elsev
```
I need to get an esp32 it looks like
Z

dsyleixa, 2020-02-08:
I actually doubt if all the defs and types in the #else...#ifndef   section will be ever needed - 
eventually all cores apart from measly 2kB AVRs have enough RAM to do without progmem anf F() and all that stuff, even the 8kB AVR Mega2560 has it.
But I personally wouldn't ever mess around with this code, I finally don't understand the crucial details over there....

dsyleixa, 2020-02-08:
simply outcommenting
```
/*        
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
*/        
```

is generating new compile errors

```
Arduino: 1.8.9 (Windows 7), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_605724 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_Test\MPU6050dmp6nIRQ\MPU6050dmp6nIRQ001\MPU6050dmp6nIRQ001.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_605724 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\arduino\portable\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose D:\Akten\Programmierung\ArduinoProgs\ESP32_Test\MPU6050dmp6nIRQ\MPU6050dmp6nIRQ001\MPU6050dmp6nIRQ001.ino
Using board 'featheresp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hard [... cut ...]
                          getFIFOBytes(Trash, (uint8_t)RemoveBytes);

                                       ^

Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: D:\arduino\portable\packages\esp32\hardware\esp32\1.0.4\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.
```

dsyleixa, 2020-02-08:
you can test this by your own, just install esp32 and choose this board, even if it's not yet plugged in.

ZHomeSlice, 2020-02-08:
`#define BUFFER_LENGTH 32`
Should do the trick. 
 This represents the i2c buffer size of your device some devices have larger buffers I googled the ESP32 and this is the size it said it has
Z

dsyleixa, 2020-02-09:
where has this to be added?
Dd you already test it then both vor ESP, ARM and old AVR cores?

ZHomeSlice, 2020-02-09:
I'm not at my home to test anything. I apologize that the support is limited 
Here is what I changed to get it to compile with a random esp32 configuration

```
#ifdef prog_void
        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
#endif
```
this should allow progmem to be used as intedded with the super cheap and small Teensy 3.0 library without causing errors to the avr boards like your ESP32
basically if Prog memory isn't used don't try to change definitions to always use it.

Next
Your error message told you to look in MPU6050.cpp
I Placed this just before the function GetCurrentFIFOPacket on line 2751

```
#ifndef BUFFER_LENGTH
// band-aid fix for platforms without Wire-defined BUFFER_LENGTH (removed from some official implementations)
#define BUFFER_LENGTH 32
#endif
```
What this does is if the definition of BUFFER_LENGTH is not present, create it and set it to 32 as a default.
Now these are pre compiler definitions like search and replace  the code from this point on is searched for the "BUFFER_LENGTH" and it is replaced with "32"  More on this here  [https://www.tutorialspoint.com/cplusplus/cpp_preprocessor.htm](https://www.tutorialspoint.com/cplusplus/cpp_preprocessor.htm)

This i2c buffer size is part of the processor itself could be different for each device but 32 seems to be a good guess. What this allows in certain conditions where the FIFO buffer is partially filled but not completely filled this allows the latest reading to quickly be retrieved by matching the exact amount of data to request that the i2c buffer can hold asking for more or less will cause additional reads across the i2c creating additional load for the MPU. While this is irrelevant for your super fast ESP32 in comparison my slow atmega328p needs every advantage that it can have to streamline this process. 

Please let me know if this works as I will suggest these changes to the library for everyone to use.
Z

dsyleixa, 2020-02-09:
thank you,
I have added the buffer length, but i do not understand what I should do where / with 

```
#ifdef prog_void
        typedef void prog_void;
        typedef char prog_char;
        typedef unsigned char prog_uchar;
        typedef int8_t prog_int8_t;
        typedef uint8_t prog_uint8_t;
        typedef int16_t prog_int16_t;
        typedef uint16_t prog_uint16_t;
        typedef int32_t prog_int32_t;
        typedef uint32_t prog_uint32_t;
#endif
```

[22 more replies]