jrowberg/i2cdevlib #444: DMP values go erratic due to interference. How to Reset Coms?

GearDownForWhat, 2019-05-28:
Hey all,
I have had this issue occur with 3 different Arduino boards, and 3 different MPU-6050's, the common denominator is _mostly_ the motor driver, but the issue can _sometimes_ occur with the motor driver not even in play. 
So, Issue is partially related to the large motor driver mounted to the same chassis, and the interference is _mostly_ related to the proximity of the motor driver to the MPU6050&Teensy

The solution that I am searching for is the ability to recover communications with the MPU-6050 without rebooting everything. 

MPU-6050, Teesny 3.6, Odrive 3.6.
Everything is wired as expected, and works most of the time, but every now and then the MPU6050 starts spitting out values that are all over the map, almost as if they were all shifted by one bit.  This only happens when the Teensy and 6050 are in the vicinity of the motor controller, and is more and more likely the longer the wires between the Teensy and MPU are. I have a routine to detect this problem and shut everything down, but I would like to have a way to reset the DMP sketch without having to manually hit the reset button. (maybe a output to pull the reset pin low 🤣... J/K)

I'm in the process of designing a custom circuit board to make the wires as short as possible, so in a couple of weeks It likely won't be happening nearly as often, but if it would help to diagnose this issue I would be happy to send lend out some hardware where the issue is repeatable, the R/C Car from this video:
https://youtu.be/k_vYA-xRXyo

intensite, 2019-07-16:
Did you found a solution to this problem?  
I am experiencing similar issues.

GearDownForWhat, 2019-07-16:
The only solution I’ve found is to keep the wires less than an inch long.

paynterf, 2019-12-12:
I'm having similar problems, and even zero-length I2C lines is not a guarantee.  See my posts at fpaynter.com ([this post](https://www.fpaynter.com/2019/07/mpu6050-imu-motor-noise-troubleshooting/), for instance, among many others).

You mentioned you were using the Teensy 3.6; can you tell me what MPU6050 libraries you are using for this?  I can't seem to get the MPU6050 libraries here to compile for the Teensy 3.x line

Frank

GearDownForWhat, 2019-12-12:
Read your post, would suggest others read it too, it is a very detailed account that matches the issue i was having exactly.  As for issue of compiling for teensy 3.6, i don't recall having that issue but working code (MPU 6050) for my robot that compiles (with a modified PID library) is uploaded on my page

paynterf, 2019-12-12:
Yeah, I've been fighting this problem a lot.  If you do a search on my site
for 'Motor Noise', you'll find several relevant posts ;-)

Hmm, I looked around a bit on your GearDownForWhat site for your
Teensy/MPU6050 code, but couldn't find it.  Did you maybe mean your GitHub
account instead?

TIA,

Frank

ZHomeSlice, 2019-12-13:
Frank,
Jumping in again :)

While I haven't run my balancing bot for hours on end. I have not experienced locking issues. I custom made my Arduino board and drive the motors off of the same 12V pack that powers the motors.
My power supply is a more powerful 5V supply than what comes with the traditional UNO and I have added 6 Capacitor scattered across the board near points of potential voltage surges critical supply points for the atmega328 chip. I also power the MPU605 from this 5V source without any modifications. My Motors are DC Brush motors. Note my motors when locked up will easily blow a 10amp automotive fuse and the motor controller will handle around 80 amps if needed @ 40amps per motor. 

Some thoughts.
so the mpu6050 breakout board uses  4.7K ohm resistors for the 3.3V Pull-up see schematic:
[https://components101.com/sites/default/files/inline-images/MPU6050-Circuit_0.png](https://components101.com/sites/default/files/inline-images/MPU6050-Circuit_0.png)

To combat magnetic interference, what if we added (2) 4.7k ohm resistors (one for data and the other for the Clock) to the Arduino Uno 3.3V supply as a supplemental source.
Thoughts
8.8K should prevent the supplies from interfering with each other as the voltage should be almost exactly the same from ground.  in this way even if one i2c pullup source receives interference the second should be good. data is transmitted by shorting the i2c lines to ground. 2.4k ohms is within the mpu-6050 i2c specs for pullup resistors which can sync 1 milliamp to ground or a minimum resistance of 1.1K ohms on the pullup resistors (see MPU-6000/MPU-6050 Product Specification page 16)

Second thought what if the locking issue is caused by RF interference. Placing the MPU6050 inside a, let's say a simple aluminum foil enclosure to isolate it could resolve the issue. 

Another possibility is power supply noise so adding additional capacitors before and or after the voltage regulators could solve the interference issues

Homer

p.s. GearDownForWhat - I love what you did with your balancing bot Nice Job!