jrowberg/i2cdevlib #448: MPU6050 compilation "error"

dmb0058, 2019-07-10:
Compiling on Ubuntu 18.04 with arduino-mk I get:

./sketchbook/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:522:65: warning: integer overflow in expression [-Woverflow]
         - (int32_t)qI[2] * qI[2] + (int32_t)qI[3] * qI[3]) / (2 * 16384);

MalcolmBoura, 2019-09-02:
Looks like that part of the expression is being evaluated in an int16_t context in which case the previous two statements may have the same problem which will result in dramatic loss of precision.

JAY-M-ERICSSON, 2020-03-06:
C++ evaluate integral constat to int, and an int on a small arduino is indeed a int16_t. 

the fix is to change  (2 * 16384) into  (2 * 16384**L**), which will force 16384 to be seen as a long, so on 4 bytes (int32_t)