jrowberg/i2cdevlib #450: "MPU6050_6Axis_MotionApps_V6_12.h" and "newly added calibration code in MPU6050.cpp" make compile error.

kmg3821, 2019-07-13:
There are mismatches for data type. For instance, in  "MPU6050_6Axis_MotionApps_V6_12.h",

I2Cdev::writeBit(devAddr,0x6B, 7, &(val = 1));  
--> Last parameter's data type is _uint8_t*

However, in "I2Cdev.h",

static bool writeBit(uint8_t devAddr, uint8_t regAddr, uint8_t bitNum, uint8_t data); 
--> It must be _uint8_t.

intensite, 2019-07-15:
Having similar problem with Arduino IDE when trying to compile MPU6050_DMP_using_DMP_V6.12

```

Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\stephen\AppData\Local\Temp\arduino_modified_sketch_788500\MPU6050_DMP6_using_DMP_V6.12.ino: In function 'void setup()':

MPU6050_DMP6_using_DMP_V6.12:217:13: error: 'class MPU6050' has no member named 'CalibrateAccel'

         mpu.CalibrateAccel(6);

             ^

MPU6050_DMP6_using_DMP_V6.12:218:13: error: 'class MPU6050' has no member named 'CalibrateGyro'

         mpu.CalibrateGyro(6);

             ^

MPU6050_DMP6_using_DMP_V6.12:220:13: error: 'class MPU6050' has no member named 'PrintActiveOffsets'

         mpu.PrintActiveOffsets();

             ^

exit status 1
'class MPU6050' has no member named 'CalibrateAccel'

```

jrowberg, 2019-07-18:
As long as you have the latest code from the master branch (particularly the `/Arduino/MPU6050/*` files), this should no longer happen.

closed 2019-07-18 by jrowberg

bart, 2020-04-03:
Same issue with ESP32 :(
```
'class MPU6050' has no member named 'PrintActiveOffsets'
```

intensite, 2020-04-03:
> Same issue with ESP32 :(
> 
> ```
> 'class MPU6050' has no member named 'PrintActiveOffsets'
> ```

What is the date of the changelog on top of your MPU6050.cpp file ?

Mine is this and it works:
```
Changelog:
//  2019-07-08 - Added Auto Calibration routine
```
I struggled a lot to make it work in a custom C++ class on a ESP32 but now it is pretty good.

bart, 2020-04-03:
Thanks a lot for your fast reply @intensite !

I moved folders `I2Cdev` and `MPU6050` from Arduino into my custom libraries folder. When executing DMP 6.12 example I'm getting `typedef unsigned long prog_uint32_t;` error for ESP32 board. So what I did was replacing custom lib files with files from `ESP32_ESP-IDF/components` which seems like they not include all necessary methods. Must I use the `ESP32_ESP-IDF` folder in a different way?

bart, 2020-04-03:
Got it working now with DMP6 demo instead. Had to do some changes like described here: https://github.com/jrowberg/i2cdevlib/pull/367 But instead of adapting `I2Cdev.cpp` change file `MPU6050.cpp` instead.

intensite, 2020-04-03:
Yes I recall seing this. I probably applied this change early on but just didn't remember.
What do you use for I2CDEV_IMPLEMENTATION? I was never able to use anything else than the standard "Wire.h" with the ESP32.   I use it like this  

```Wire.begin(SDA, SCL, 400000);```

bart, 2020-04-03:
It uses the default here for I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE which is just a `Wire.begin();`