jrowberg/i2cdevlib #456: Code changes for MPU6050 driver cause compile errors for ESP8266

johnhmacleod, 2019-07-17:
Latest changes using default args cause multiple compile failures for the ESP8266 platform.

```
C:\Program Files (x86)\Arduino\libraries\MPU6050\MPU6050.cpp:3215:46: error: default argument given for parameter 1 of 'void MPU6050::CalibrateGyro(uint8_t)' [-fpermissive]

 void MPU6050::CalibrateGyro(uint8_t Loops = 6) {

                                              ^

In file included from C:\Program Files (x86)\Arduino\libraries\MPU6050\MPU6050.cpp:38:0:
```

alexthe-red, 2019-07-18:
Using the Arduino IDE for multiple boards yields similar error:

`/Users/redacted/Documents/Arduino/libraries/MPU6050/MPU6050.cpp:3215:6: error: prototype for 'void MPU6050::CalibrateGyro(uint16_t)' does not match any in class 'MPU6050'
 void MPU6050::CalibrateGyro(uint16_t Loops = 6) {`

jrowberg, 2019-07-18:
Just committed a few changes that should take care of both of these issues (as long as you have all of the latest `/Arduino/MPU6050/*` files from the master branch).

alexthe-red, 2019-07-18:
That fixed my issue, thanks!