jrowberg/i2cdevlib #457: MPU6050.cpp - error: 'dtostrf' was not declared in this scope

nhphuong91, 2019-07-23:
Hi, I got this error when compile my sketch on VSCode using all latest libraries here.
The full error output is quite lengthy so let me summarize it:
The is an error: `MPU6050/MPU6050.cpp:3307:196: error: 'dtostrf' was not declared in this scope` which happen with the macro `printfloatx`

ZHomeSlice, 2019-07-25:
dtostrf  = Displaying FLOAT variables
this converts a float to a string
The printfloatx() is a macro that is just a fancy way to print floats I have a fixes space buffer to insert the float into with an easy way to handle text before and after all in one macro.
Temporary fix:
Just remove the printfloatx(), and replace it with Serial.print();

also what platform are you using it on? 
I thought the dtostrf() function was a standard part of of the Arduino UNO libraries.

ZHomeSlice

> Arduino Float to String https://dereenigne.org/arduino/arduino-float-to-string/
> Sat Mar 5, 2011 by jmccrohan in  Arduino  Arduino, double, dtostrrf, float, sprintf, string
> 
> If you have ever tried to use sprintf() on an Arduino to convert from a float to a string, you will notice it doesn’t work.
> 
> sprintf(buf,"%f", floatvar);
> The above function will most likely return a “?” to your char buffer.
> 
> If you google around, you’ll see various functions that people have written to overcome this, but all of them seem broken in one way or another. The alternative is to use dtostrf(), a standard avr-libc function. This provides a simple, tested function to convert from a float to a string.
> 
> To use this function, simply replace sprintf() with dtostrf(), along with the self explanatory parameters below.
> 
> dtostrf(floatVar, minStringWidthIncDecimalPoint, numVarsAfterDecimal, charBuf);
> As the function name suggests, this will also work for the conversion of doubles to strings.

nhphuong91, 2019-07-26:
@ZHomeSlice Yeah, I think it's because Due uses ARM core instead of regular atmega chip. Thus it has different std lib than other avr boards.

ZHomeSlice, 2019-07-27:
replace my #define with one of these:
```
#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) print(Name); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(Variable);}Serial.print(EndTxt);//Name,Variable,Spaces,Precision,EndTxt
//#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) print(Name); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(sprintf( s, "%*.*f", width, prec, Variable ));}Serial.print(EndTxt);//Name,Variable,Spaces,Precision,EndTxt
```
I know the first one will work I'm guessing on the second one but it is an example for you to go off of
The first one doesn't handle spacing at all numbers will be placed one after the other  you may need to adjust for this with extra spaces as needed.

sabas1080, 2019-09-05:
for SAMD add
#include <avr/dtostrf.h>

and work

nhphuong91, 2019-09-16:
I found another solution, credit to nestachan [here](https://forum.arduino.cc/index.php?topic=140695.0)

> Adding header file  <avr/dtostrf.h> to WString.h file.
Path: /home/<user>/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino

closed 2020-09-03 by nhphuong91

quazi-irfan, 2021-01-31:
@ZHomeSlice 's answer is the most sensible solution here.