jrowberg/i2cdevlib #458: MPU605 DMP support for STM32

BronantheBrobarian, 2019-07-23:
Does anyone have an idea about when the STM32 lib will support DMP?
Is it under development currently?
Any information is greatly appreciated
Thank you

Pluscrafter, 2019-07-23:
I have an working example with the STM32 HAL library(created with CubeMX). I'll post the code this evening you just have to change the spi port and the pins for the library. It DMP works but the code is a little tidy.

Pluscrafter, 2019-07-23:
[https://github.com/Pluscrafter/i2cdevlib/tree/master/STM32_HAL/Nucleo-144F722ZE](url)

BronantheBrobarian, 2019-07-25:
Hi, thanks for the reply!
Would really appreciate taking a look at the code.
I don't think your link is working though.
I am using the Sparkfun Pro nRF52840 mini dev board and I wanted to use the DMP on the MPU6050.

Pluscrafter, 2019-07-26:
New link

https://github.com/Pluscrafter/i2cdevlib/tree/master/STM32_HAL/Nucleo-144F722ZE

josb86, 2020-02-17:
> New link
> 
> https://github.com/Pluscrafter/i2cdevlib/tree/master/STM32_HAL/Nucleo-144F722ZE
> 
> I'm using a Nucleo F722ZE board.

Hi, I need you help whhen I build the project in stm32CubeIde, this show 2 errors, the frist is:

Description	Resource	Path	Location	Type
fatal error: MPU6050_.h: No such file or directory	stm32f7xx_it.cpp	/Nucleo-144F722ZE/Src	line 43	C/C++ Problem

Description	Resource	Path	Location	Type
make: *** [Src/subdir.mk:66: Src/stm32f7xx_it.o] Error 1	Nucleo-144F722ZE		 	C/C++ Problem

Then I change #include "MPU6050_.h" for #include "MPU6050.h" in stm32f7xxit.cpp

and build again  and get another error:

Description	Resource	Path	Location	Type
'Sensors' has not been declared	stm32f7xx_it.cpp	/Nucleo-144F722ZE/Src	line 231	C/C++ Problem

if(HAL_GPIO_ReadPin(GPIOC,GPIO_PIN_6)){ // @suppress("Invalid arguments") // @suppress("Symbol is not resolved")
	  Sensors::MPU6050::dataRDY = true;
  }

What is worng? I need run this on a nucleo-f746ZG.