jrowberg/i2cdevlib #460: DMP of MPU6050 only shows quaternion´s components as zeros and fifo is overflown

misha717, 2019-07-26:
Hi everybody.
I am trying to use the dmp of the mpu 6050. 
In order to do this, I am using the sketch "MPU6050_DMP6.ino" with almost no changes.

When I run the sketch the dmp initializes correctly, but then, when it shows the quaternion´s components it only shows zeros, besides the fifo is overflown once evey 14 printed values.
Similar problems occur when i try to get euler angles or yaw, pitch and roll.

Any idea of what may be happening?
Thanks a lot!
Below is the code I am using.
```C

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

/* =========================================================================
   NOTE: In addition to connection 3.3v, GND, SDA, and SCL, this sketch
   depends on the MPU-6050's INT pin being connected to the Arduino's
   external interrupt #0 pin. On the Arduino Uno and Mega 2560, this is
   digital I/O pin 2.
 * ========================================================================= */

/* =========================================================================
   NOTE: Arduino v1.0.1 with the Leonardo board generates a compile error
   when using Serial.write(buf, len). The Teapot output uses this method.
   The solution requires a modification to the Arduino USBAPI.h file, which
   is fortunately simple, but annoying. This will be fixed in the next IDE
   release. For more info, see these links:
   http://arduino.cc/forum/index.php/topic,109987.0.html
   http://code.google.com/p/arduino/issues/detail?id=958
 * ========================================================================= */

// uncomment "OUTPUT_READABLE_QUATERNION" if you want to see the actual
// quaternion components in a [w, x, y, z] format (not best for parsing
// on a remote host such as Processing or something though)
#define OUTPUT_READABLE_QUATERNION

// uncomment "OUTPUT_READABLE_EULER" if you want to see Euler angles
// (in degrees) calculated from the quaternions coming from the FIFO.
// Note that Euler angles suffer from gimbal lock (for more info, see
// http://en.wikipedia.org/wiki/Gimbal_lock)
//#define OUTPUT_READABLE_EULER

// uncomment "OUTPUT_READABLE_YAWPITCHROLL" if you want to see the yaw/
// pitch/roll angles (in degrees) calculated from the quaternions coming
// from the FIFO. Note this also requires gravity vector calculations.
// Also note that yaw/pitch/roll angles suffer from gimbal lock (for
// more info, see: http://en.wikipedia.org/wiki/Gimbal_lock)
//#define OUTPUT_READABLE_YAWPITCHROLL

// uncomment "OUTPUT_READABLE_REALACCEL" if you want to see acceleration
// components with gravity removed. This acceleration reference frame is
// not compensated for orientation, so +X is always +X according to the
[... 274 lines cut ...]
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

```

ZHomeSlice, 2019-07-27:
Add this line of code at the         
"// other program behavior stuff here" section
       
```
  static unsigned long _ETimer; // My exact timer (Exactly 10 ms apart)
  if ( millis() - _ETimer >= 10) {
    _ETimer += 10;
    mpuInterrupt = true; // Let's trigger a check of the MPU even though we have not received an interrupt
  }
```
Let me know if this solves your problem. If it does your interrupt trigger is not properly functioning.
This adds a little overhead to the program but it should solve the problem if it is an interrupt issue.
this is a good way to eliminate the interrupt pin connection if that is an issue

misha717, 2019-07-28:
Thanks for your interest zHomeSlice, I really apreciate your help! 👍 
I added the code you posted in the script but the same problem keeps happening.

ZHomeSlice, 2019-07-28:
In I2Cdev.h has a debug print feature. This should display every transaction between the MPU and UNO. Enable the debug feature and post the output. 
The code works on all my mpu's.

You may consider posting on the Arduino forum and place a link here and I'll jump in. I believe this isn't a direct library issue.
I'm suspecting this to be hardware or wiring issue.

misha717, 2019-07-30:
https://arduino.stackexchange.com/questions/67466/dmp-of-mpu6050-only-shows-zeros-and-fifo-is-overflown

misha717, 2019-07-30:
The output is very long, and i can´t post everything. Here I leave some parts that I belive are important. 

Initializing I2C devices...
I2C (0x68) reading 1 bytes from 0x6B...1. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...1. Done.
I2C (0x68) reading 1 bytes from 0x1B...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x1B...0. Done.
I2C (0x68) reading 1 bytes from 0x1C...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x1C...0. Done.
I2C (0x68) reading 1 bytes from 0x6B...1. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...1. Done.
I2C (0x68) reading 1 bytes from 0x6B...1. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...1. Done.
Testing device connections...
I2C (0x68) reading 1 bytes from 0x75...68. Done (1 read).
MPU6050 connection successful

Send any character to begin DMP programming and demo: 

Codigo comentado de los 3 ciclos while 
Initializing DMP...
I2C (0x68) reading 1 bytes from 0x6B...1. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...81. Done.
I2C (0x68) reading 1 bytes from 0x6B...40. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...0. Done.
I2C (0x68) writing 1 bytes to 0x6D...70. Done.
I2C (0x68) writing 1 bytes to 0x6E...6. Done.
Checking hardware revision...
Revision @ user[16][6] = I2C (0x68) reading 1 bytes from 0x6F...4D. Done (1 read).
4D
Resetting memory bank selection to 0...
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x25...7F. Done.
I2C (0x68) reading 1 bytes from 0x6A...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6A...0. Done.
I2C (0x68) writing 1 bytes to 0x25...68. Done.
I2C (0x68) reading 1 bytes from 0x6A...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6A...2. Done.
I2C (0x68) reading 1 bytes from 0x6B...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x6B...3. Done.
I2C (0x68) writing 1 bytes to 0x38...12. Done.
I2C (0x68) writing 1 bytes to 0x19...4. Done.
I2C (0x68) reading 1 bytes from 0x1A...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x1A...8. Done.
I2C (0x68) reading 1 bytes from 0x1A...8. Done (1 read).
I2C (0x68) writing 1 bytes to 0x1A...B. Done.
I2C (0x68) reading 1 bytes from 0x1B...0. Done (1 read).
I2C (0x68) writing 1 bytes to 0x1B...18. Done.
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...0. Done.
I2C (0x68) writing 16 bytes to 0x6F...FB 0 0 3E 0 B 0 36 0 1 0 2 0 3 0 0. Done.
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...0. Done.
I2C (0x68) reading 16 bytes from 0x6F...FB 0 0 3E 0 B 0 36 0 1 0 2 0 3 0 0. Done (16 read).
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...10. Done.
I2C (0x68) writing 16 bytes to 0x6F...0 65 0 54 FF EF 0 0 FA 80 0 B 12 82 0 1. Done.
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...10. Done.
I2C (0x68) reading 16 bytes from 0x6F...0 65 0 54 FF EF 0 0 FA 80 0 B 12 82 0 1. Done (16 read).
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...20. Done.
I2C (0x68) writing 16 bytes to 0x6F...0 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done.
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...20. Done.
I2C (0x68) reading 16 bytes from 0x6F...0 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (16 read).
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
I2C (0x68) writing 1 bytes to 0x6E...30. Done.
I2C (0x68) writing 16 bytes to 0x6F...0 28 0 0 FF FF 45 81 FF FF FA 72 0 0 0 0. Done.
I2C (0x68) writing 1 bytes to 0x6D...0. Done.
[... 92 lines cut ...]
I2C (0x68) reading 2 bytes from 0x72...1 50. Done (2 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
quat	0.00	0.00	0.00	0.00
I2C (0x68) reading 1 bytes from 0x3A...3. Done (1 read).
I2C (0x68) reading 2 bytes from 0x72...1 F8. Done (2 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
[... cut ...]
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
I2C (0x68) reading 42 bytes from 0x74...0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0. Done (42 read).
quat	0.00	0.00	0.00	0.00

misha717, 2019-07-30:
I changed the sensor and now it works.
Thanks for your help.

closed 2019-07-30 by misha717