jrowberg/i2cdevlib #461: MPU6050 calibration code using random/undetermined values

neorevx, 2019-07-27:
I don't know if the intent is to use a random/undetermined value for the "Reading" variable, but this variable in the code below (lines 3265 and 3267) is not initialized, so its value is undetermined.

https://github.com/jrowberg/i2cdevlib/blob/97c703909f209710c6387d7a2cb5f50ceaaab496/Arduino/MPU6050/MPU6050.cpp#L3250-L3269

ZHomeSlice, 2019-07-27:
Thank  you,
Nice Catch.
I missed this after I simplified my reworking of the PID function. 
The intent is to store Data into Reading: Reading (float) == Data (int) 
While the PID loop will find the offsets either way. This should drastically improve the discovery time of the offsets if we don't start with a random number!
insert at line 3264:
`		Reading = Data;`
Z

ZHomeSlice, 2019-07-28:
Issue is fixed with [pull 462](https://github.com/jrowberg/i2cdevlib/pull/462)