jrowberg/i2cdevlib #466: MPU6050 Documentations

i3130002, 2019-08-06:
I got involved in a MPU6050 project, honestly this one really hearts! No well known documentation and not having an error free code anywhere

> https://www.i2cdevlib.com/devices/mpu6050#source
> Due to the lack of good publicly available documentation about the inner workings of this device, all DMP-related information has been reverse-engineered from I2C signal analysis correlated against the company-provided Embedded MotionApps software

I am willing to share what I gathered over the past month or so. (Some might seems to be duplicates but they have minor differences, revision 4.2, for example is missing DMP control registers!!! ) 
I also include sample codes provided by the manufacturer which you can download as instructed in [here  ](https://twitter.com/Mahdi_Raf/status/1155766579793137664)

Full Docs and provided codes are available at [here](https://mega.nz/#F!wlZCRKLb!2a3xr0MLz-ZcEmSe6taekg)

ZHomeSlice, 2019-08-08:
Thank you, I've registered with Invensense years ago. and this while helpful I've struggled to have a functioning code on an Arduino UNO (atmega328p) as is. As you know there are 2 parts to the MPU. everyone has some working code for the registers and can get data from the MPU quickly. What is more difficult is gathering data from the DMP firmware. The Firmware registers change between instances and the documentation is vague as you mentioned. [My project Simple_MPU6050](https://github.com/ZHomeSlice/Simple_MPU6050) took a different approach to solve the problem. I changed all the DMP registers using several sketches. after determining which provided the best results I re-captured the firmware instance with registers changed prior to executing it. this way it is pre-configured for a specific purpose although it could be re-configured to do something more/else. most of us here in the Arduino community could care less about things like android orientation etc. 
instead of creating a function for each operation. I still use Jeffs I2Cdev library but I created macros for each register and even bits to simplify the final code. place the macro in your code and it executes a common function using I2CDev library that is configured to perform the proper register/bit read/write. 

I am interested in implementing the 9 Deg of freedom by using the magnetometer as part of the equation. It looks to be a separately compiled library the Arduino UNO must execute. I am hoping to use that library but my skills at programming are maxing out as this is a hobby for me. :)
Z

i3130002, 2019-08-09:
I finally acknowledge that this library is the most complex one out there but during my experiments I noticed that is is not that well designed and even has some critical flaws in it. The provided manufacturer files goes deep on how to compile DMP thus I thought there might be some source code in it to, the goal was to give others a chance to figure it out as I might not be able to do so.

ZHomeSlice, 2019-08-09:
> I also include sample codes provided by the manufacturer which you can download as instructed in [here](https://twitter.com/Mahdi_Raf/status/1155766579793137664)

So Go to this directory int the 6.12 download you linked to 
C:\...\motion_driver_6.12\msp430\eMD-6.0\core\mpl
in this directory, there is a precompiled library with a bunch of header files.
libmplmpu.lib
This is the holy grail 9 deg of freedom library. It will integrate the DMP with the Magnetometer readings to provide the 9th degree of freedom we have all been drooling over!!!

Can this library work on an atmega MPU. if so can you provide me with a compiling sketch to test?
Z

i3130002, 2019-08-09:
Will this help ? motion_driver_6.12\msp430\eMD-6.0\core\driver\eMPL\inv_mpu.c it seems that it has compass support! 
I do not have the micro you are asking for nor the compass module to test things.

ZHomeSlice, 2019-08-09:
I can access the compass and I have DMP mostly figured out.
To access the compass you can just treat the compass as an external i2c device. the DMP doesn't use the compass at all. nothing I can see allows compass data to be passed to the DMP. As far as I can tell mixing the 6 degrees of freedom data from the DMP with the Magnetometer data from the compass requires the use of another MPU like the atmega328p. The Invensence library libmplmpu.lib is proprietary and it contains compass calibration among other things.

> The MPL
The Motion Driver 6.12 contains a binary library which contains InvenSense proprietary algorithms for sensor fusion and dynamic calibration. The MD 6.12 driver pushes the sensor data into the MPL and the MPL will handle the 9-axis sensor fusion including the compass integration.

Reference: Motion Driver 6.12 – Features User Guide

Z

i3130002, 2019-08-09:
> Z

I never had to do so, in case I saw something useful will let you know. for now take a look at [this wiki](https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion)

i3130002, 2019-08-25:
I was fidgeting around an saw this 
[image]

In case this is what you(@ZHomeSlice ) were looking for take a look at `motion_driver_6.12\arm\STM32F4_MD6\Projects\eMD6\core\driver\eMPL\inv_mpu.c` to me it seems like current DMP headers are just able to make sense of either MPU6050 with AK8975_SECONDARY or MPU6500 with AK8963_SECONDARY.
Take a look and tell me if it was useful or not.

ZHomeSlice, 2019-08-25:
MPU 6050 with compass is a MPU9150
MPU6500 with compass is a MPU9250
Better compass in the MPU9250 
Slight difference in programming
The code to integrate magnetometer data with the 6 degrees of freedom is still illusive.

i3130002, 2019-08-26:
> MPU 6050 with compass is a MPU9150
> MPU6500 with compass is a MPU9250
> Better compass in the MPU9250
> Slight difference in programming
> The code to integrate magnetometer data with the 6 degrees of freedom is still illusive.

You know that when they are using another chip for magnetometer they are forced to use i2c, so what stops you from using one ?

i3130002, 2019-08-28:
> motion_driver_6.12\msp430\eMD-6.0\core\mpl from this website Requires registration https://www.invensense.com/developers/login/?redirect_to=software-downloads) to integrate the

The other day I was looking around and noticed EMD v5.1.3 And I noticed that they quote "`Embedded Motion Driver 5.1.3 is our six-axis version of eMD 6.1. If a compass in not required for your application, than we recommend using this`" thus there is no ready to use solution in case you are using others magnetometer. In this case the only way remains is to disassemble DMP code and change i2c registers to your needs, though it would not initialize it by default if it has some configuration, you might need to do it by micro controller or doing way more low level code editing.
Does it make sense ?

ZHomeSlice, 2019-08-28:
Yes, it just confirms that DMP has no obvious way to integrate the compass. Which if you peal the example code apart you discover compass intercession is not part of dmp but requires you to use a library. that they have created for specific processor types. if we can figure out how to make this library work with some or all the Arduino products it would provide compass integration