jrowberg/i2cdevlib #467: MPU6050::dmpGetQuaternion Wrong scale divider!

i3130002, 2019-08-09:
I was reading **MPU6050_6Axis_MotionApps_V6_12.h** and noticed that something is off by a lot!
```
uint8_t MPU6050::dmpGetQuaternion(Quaternion *q, const uint8_t* packet) {
    // TODO: accommodate different arrangements of sent data (ONLY default supported now)
    int16_t qI[4];
    uint8_t status = dmpGetQuaternion(qI, packet);
    if (status == 0) {
        q -> w = (float)qI[0] / 16384.0f;
        q -> x = (float)qI[1] / 16384.0f;
        q -> y = (float)qI[2] / 16384.0f;
        q -> z = (float)qI[3] / 16384.0f;
        return 0;
    }
    return status; // int16 return value, indicates error if this line is reached
}
```
According to the documents Gyroscope Sensitivity scale is from 16.4 to 131 
[image]

Where as Accelerameter is 2048 to 16384 
[image]

As far as I know Quaternions are for rotation and not acceleration; Can someone please explain to me why it's dividing by acceleration scales to get rotation ?

ZHomeSlice, 2019-08-10:
> As far as I know Quaternions are for rotation and not acceleration; Can someone please explain to me why it's dividing by acceleration scales to get rotation?

The Accelerometer sensing Gravity is used to calculate pitch and roll. I am not sure how the DMP math works and how much the Gyro plays a part with the pitch and roll. I'm sure it does where it is difficult to calculate these with the accelerometer alone.

Also, I believe the conversion you are pointing out has nothing to do with scale, is just shifting the decimal point of the integer to make it a floating-point number near 1.0.
Quaternation values are near 1 and so an integer has to be converted to a floating-point number.
` q -> w = (float)qI[0] / 16384.0f;`
‭0100000000000000 BIN‬ = 16384 DEC
0000000000000001B * 16384 = ‭0100000000000000 BIN‬ 
so:
Signed integer 0111 1111 1111 1111B (32,768) convert it to float = (0(sign)10001110(exponent)00000000000000000000000)
/ 16384  = 2.0 or (0(sign)10000000(exponent)00000000000000000000000)
Try the math out: [floating-point converter](https://www.h-schmidt.net/FloatConverter/IEEE754.html)
The resulting math is ‭similar to shifting bits `q -> w =  qi[0] >> 14 ` but it doesn't scrap the bits as they are shifted right  thus converting an integer to a usable floating-point number in our case near 1.0 for Qaternions.
z

i3130002, 2019-08-11:
Why is that you have to use 16384 ? How can you know qI[0] is this big (I wasn't able to find out where dmpPacketBuffer get filled) ?

ZHomeSlice, 2019-08-11:
This is somewhat difficult to discover but it is there. most of the code is written in 32 bit so the divider is much larger 2^30 or 1000000000000000000000000000000 Bin or 1,073,741,824
instead of 2^14 or 100000000000000 Bin or 16384 for 16-bit data.

Go to this directory: **\motion_driver_6.12\msp430\eMD-6.0\core\mllite\**
file: **data_builder.c** starting at line **762**
```
* @param[in] quat Quaternion data. 2^30 = 1.0 or 2^14=1.0 for 16-bit data. 
*                 Real part first. Length 4.
```
2^14 = 16384 (for 16-bit data)

Most of the math is done in 32 bit which would be 2^30.
This leaves 1 bit for the sign and 1 bit / 0 just like the 16-bit data

results_holder.c line 54
`/** @internal
* Store a quaternion more suitable for gaming. This quaternion is often determined
* using only gyro and accel.
* @param[in] quat Length 4, Quaternion scaled by 2^30 
*/
void inv_store_gaming_quaternion(const long *quat, inv_time_t timestamp)
{
    rh.status |= INV_6_AXIS_QUAT_SET;
    memcpy(&rh.gam_quat, quat, sizeof(rh.gam_quat));
    rh.gam_timestamp = timestamp;
}`

Z

ZHomeSlice, 2019-08-11:
> I wasn't able to find out where dmpPacketBuffer get filled

You can either use the MPU to fill the FIFO  **OR** the DMP firmware instance to fill the FIFO you can't use both.
The code here in the MPU6050 uses the DMP because you can't get Quaternion values any other way

Go to this directory: *\motion_driver_6.12\msp430\eMD-6.0\core\driver\eMPL\*
in file: inv_mpu_dmp_motion_driver.c line 985
```
/**
 *  @brief      Enable DMP features.
 *  The following \#define's are used in the input mask:
 *  \n DMP_FEATURE_TAP
 *  \n DMP_FEATURE_ANDROID_ORIENT
 *  \n DMP_FEATURE_LP_QUAT
 *  \n DMP_FEATURE_6X_LP_QUAT
 *  \n DMP_FEATURE_GYRO_CAL
 *  \n DMP_FEATURE_SEND_RAW_ACCEL
 *  \n DMP_FEATURE_SEND_RAW_GYRO
 *  \n NOTE: DMP_FEATURE_LP_QUAT and DMP_FEATURE_6X_LP_QUAT are mutually
 *  exclusive.
 *  \n NOTE: DMP_FEATURE_SEND_RAW_GYRO and DMP_FEATURE_SEND_CAL_GYRO are also
 *  mutually exclusive.
 *  @param[in]  mask    Mask of features to enable.
 *  @return     0 if successful.
 */
int dmp_enable_feature(unsigned short mask)...
```
we have enabled DMP_FEATURE_6X_LP_QUAT in the same manner as they do here. by changing the value in the DMP firmware located at CFG_8 which if you look up at the top of the file 
#define CFG_8                   (2718)
so at memory location 2718 in the DMP image we are writing 4 bytes:
DINA20 = 0x20
DINA28 = 0x28
DINA30 = 0x30
DINA38 = 0x38
see file: dmpkey.h in the same directory

```
/**
 *  @brief       Generate 6-axis quaternions from the DMP.
 *  In this driver, the 3-axis and 6-axis DMP quaternion features are mutually
 *  exclusive.
 *  @param[in]   enable  1 to enable 6-axis quaternion.
 *  @return      0 if successful.
 */
int dmp_enable_6x_lp_quat(unsigned char enable)
{
    unsigned char regs[4];
    if (enable) {
        regs[0] = DINA20;
        regs[1] = DINA28;
        regs[2] = DINA30;
        regs[3] = DINA38;
    } else
        memset(regs, 0xA3, 4);

    mpu_write_mem(CFG_8, 4, regs);

    return mpu_reset_fifo();
}
```
How do we know we are accessing the DMP instead of just a register? We are using the same function we used to set the DMP into the MPU6050
mpu_write_mem() only accesses 2 registers in the MPU to do everything with the DMP 
bank_sel register = 0x6D
and mem_r_w register 0x6F

Z

i3130002, 2019-08-12:
I hate it when I'm wrong, although this time there was not way I could have figure it by myself! Thank you so much for your time and effort.
I exported the raw data(rotating as fast as I could around y axis!) into excel and got the following results:
[image]
From above one can conclude that 
qI[0] had highest values, 
Max is very close to the `16384.0f` in question,
using Quaternion calculator like [this one](https://www.andre-gaschler.com/rotationconverter/) we get the following which is very close 
[image]

closed 2019-08-12 by i3130002