jrowberg/i2cdevlib #468: MPU6050_6Axis_MotionApps20.h implementation in .h file cause problems

intensite, 2019-08-09:
The file MPU6050_6Axis_MotionApps20.h contains BOTH definitions AND implementations.

This cause problems when included in a class .h file and a class definition file .cpp see my class definition bellow.

```c++
// File: gyro.h
#include "MPU6050_6Axis_MotionApps20.h"  // Must be included for the MPU6050 mpu property.

class Gyro {
    private:
        MPU6050 mpu;

        // MPU control/status vars
        bool dmpReady = false;  // set true if DMP init was successful
        uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
        uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
        uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
        uint16_t fifoCount;     // count of all bytes currently in FIFO
        uint8_t fifoBuffer[64]; // FIFO storage buffer

        // orientation/motion vars
        Quaternion q;           // [w, x, y, z]         quaternion container
        VectorInt16 aa;         // [x, y, z]            accel sensor measurements
        VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
        VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
        VectorFloat gravity;    // [x, y, z]            gravity vector
        float euler[3];         // [psi, theta, phi]    Euler angle container
        float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

    public:
        Gyro();
        void setupGyro();
        void ProcessGyroData();
};
```
 Should the definition and the implementation be separated in different files?  Otherwise do you have a suggestion on how I should proceed to use the MPU-6050 library in a c++ class set of .h and .cpp files ?

Thanks for your great work.

MArimont3, 2019-09-22:
I have exactly the same problem. The (simple) examples does work, but as soon as I try to use the library is a class, I get tons of errors : "multiple definition of MPU6050::xxxxxxxxxxxxxxxxxxxx", one for each MPU6050 class method defined in MPU6050_6Axis_MotionApps20.h
No idea what is going wrong, I am not a C++ specialist, but I fear that adding (MPU6050) class method in a *.H file, while declaring them in the MPU6050.h file is not very clean ?
Anyone an idea how to solve this ???

Bascy, 2019-10-18:
Same problem here, implementations of functions like `uint8_t MPU6050::dmpGetAccel(int32_t *data, const uint8_t* packet)` should be moved to a `MPU6050_6Axis_MotionApps20.cpp` file i guess

intensite, 2019-10-18:
To work around the issue, I placed both the definition AND implementation of my class in its .h file.
Not the most elegant solution indeed but at least I was able to make it work until a permanent fix is provided.

See: https://github.com/intensite/rocketcontrol/blob/master/src/gyro/Cgyro.h

Maclain272, 2020-12-23:
**maclain272**
Hi world !
 Hello humans!
 Allow me in for realization of self balancing robot!

jorritsmit, 2021-01-09:
i have the same issue. are you open for a pr?

jrowberg, 2021-09-26:
This should hopefully be addressed in the next week or so with a permanent fix. Since there are numerous issues mentioning it, I’m closing out all the duplicates in the meantime.

closed 2021-09-26 by jrowberg