jrowberg/i2cdevlib #471: Tiva C tm4c123 with mpu6050 stop on initailzing i2c devices...

NyHunor, 2019-08-24:
Hi good day everybody! I am trying to connect my tiva c tm4c123 with mpu6050  as in( Learning-Robotics-using-Python-Second-Edition) written. I put back ubuntu 16.04 for this "job". and followed kinda everything.. but this is where everything stops , when tiva c supposed to read raw data from mpu6050, i tried several kind of wiring (now with the original back (vcc to +3.3v     gnd to gnd   SCL to PD0     SDA to PD1)  and already tried with several version of the Energia IDE compiler even using all link from the book, what actually lead to download this i2cdev lib for arduino. so for some reason it doesn`t do the job currently i tried with Energia 1.6.10E18 but tried from E14 to the latest. 

the code is simple enough as you will see below, the IDE is checking and compiling /uploading, then i Click on to the serial monitor, setting the correct baud rate, then it stuck on initailizing i2c device even when i reset is, 
CAN YOU PLEASE ADVICE ME ! what am i making wrong, how to fix? i was looking for answer for days, with no luck. funny thing when i tried to get the address of the mpu6050 ,instead of giving me that,it gives all possible address for the tiva c card...

#include "Wire.h" 
 
#include "I2Cdev.h" 
#include "MPU6050.h" 
 
MPU6050 accelgyro;
void setup() 
{ 
   
  //Init Serial port with 115200 buad rate 
  Serial.begin(115200);   
  Setup_MPU6050(); 
} 
void Setup_MPU6050() 
{ 
   Wire.begin(); 
   
      // initialize device 
    Serial.println("Initializing I2C devices..."); 
    accelgyro.initialize(); 
 
    // verify connection 
    Serial.println("Testing device connections..."); 
    Serial.println(accelgyro.testConnection() ? "MPU6050 connection successful" : "MPU6050 connection failed"); 
} 
void loop() 
{ 
 
   //Update MPU 6050 
    Update_MPU6050();
}
void Update_MPU6050() 
{ 
   
  int16_t ax, ay, az; 
  int16_t gx, gy, gz; 
 
      // read raw accel/gyro measurements from device 
    accelgyro.getMotion6(&ax, &ay, &az, &gx, &gy, &gz); 
 
    // display tab-separated accel/gyro x/y/z values 
    Serial.print("i");Serial.print("t"); 
    Serial.print(ax); Serial.print("t"); 
    Serial.print(ay); Serial.print("t"); 
    Serial.print(az); Serial.print("t"); 
    Serial.print(gx); Serial.print("t"); 
    Serial.print(gy); Serial.print("t"); 
    Serial.println(gz); 
    Serial.print("n"); 
}

ZHomeSlice, 2019-08-25:
vcc to +3.3v

The mpu6050 breakout has a 3.3v regulator and the i2c pull up resistor is tied to the 3.3v side. If you supply 3.3v to the vcc the output is now even less to the i2c bus and the mpu6050, consider bypassing the voltage regulator or supplying the mpu6050 with 5v

NyHunor, 2019-08-28:
i just did it  and connected INT as well.. and THANK YOU it is working now! it looks like you supposed to write a book about building robots,because the aforementioned book is teeming with these kind of "mishaps"