jrowberg/i2cdevlib #476: MPU6050 not giving an int signal

yo90bosses, 2019-10-02:
Im using an esp32 with a mpu6050 and all the motionapp example sketches dont work because the mpu6050 is not sending an interrupt signal from the int pin.

yo90bosses, 2019-10-07:
I removed the interrupt variables in the DMP example sketch and now it is running without interrupts. One more problem though is that if the mpu is started in other angles than flat the angles are zero and seem to not work correctly.

ZHomeSlice, 2019-10-12:
If Calibration is on you must start it flat.
use the numbers to fix the calibration offsets into your source code and then turn off calibration.
The calibration routine relies on the assumption that gravity directly is below the MPU and so when you calibrate for accuracy you remove gravity from the acceleration to see what the error in the sensor is. this error is removed from the MPU when done. if gravity is not below the MPU's z-axis when calibration occurs then the calibration values are false and will produce incorrect readings until corrected. 
Z