jrowberg/i2cdevlib #478: MPU6050_DMP Connecting Failed

Wfim, 2019-10-08:
Hi there, I am relatively new to Arduino and etc 
I have currently set up my Arduino Uno and MPU 6050 Accelerometer & Gyroscope. 
I am attempting to run the MPU6050_DMP Program so I can run the processing Teapot simulation as I plan to use this for further projects.
I have run two other codes that have given me Accelerometer data & Gyroscope data in all 3 of their Axis. Also the temperature. I went to the Serial Monitor after uploading the program and it gave me this 

*Initializing I2C devices...
Testing device connections...
MPU6050 connection failed

Send any character to begin DMP programming and demo: 

I have tried resoldering the SDA and SCL ports on the chip. 
Also, looked around for any possible solution but have found none to my avail.
Issue attached.
Also when I have verified the file it has given me lines of orange text and I have attached another image for that although I think it is unrelated.

Any help will be grateful thanks

[image]

[image]

jc508, 2019-10-19:
looks like 'Connection Failed' simply queries the device_id (from the WHO AM I Register 0x75.  My chip answers with 0x72 not the expected 0x34.  Just ignore the 'issue' (or comment it out) and it should work

H1X4Dev, 2019-11-28:
I have the same issue here as well.

`getDeviceID()` responds with 0x77 and by ignoring testConnection() check it would just respond with the following:
```
Resetting MPU6050...
Checking hardware revision...
Revision @ user[16][6] = 77
Resetting memory bank selection to 0...
Reading OTP bank valid flag...
OTP bank is invalid!
Setting slave 0 address to 0x7F...
Disabling I2C Master mode...
Setting slave 0 address to 0x68 (self)...
Resetting I2C Master control...
Setting clock source to Z Gyro...
Setting DMP and FIFO_OFLOW interrupts enabled...
Setting sample rate to 200Hz...
Setting external frame sync to TEMP_OUT_L[0]...
Setting DLPF bandwidth to 42Hz...
Setting gyro sensitivity to +/- 2000 deg/sec...
Writing DMP code to MPU memory banks (1929 bytes)
-> Initial memory load failed (dmgInitialize() result: 1).
```

I wonder if it has something to do with the chip itself, I tried the others in the package as well, but all would yield the same result. Despite being GY-521, would it be possible for manufacturer error?

Pradhansuman3112, 2020-11-25:
i don't understand please clarify briefly