jrowberg/i2cdevlib #484: Warnings at compile time

jc508, 2019-10-19:
Hi,
Compiling in the Arduino IDE yields over 1000 lines of warnings.
Apart from being 'noise' it looks like some are generated by multiple definitions of objects.
Am I unusual or does everybody see these ??
[MPU6050_DMP6_compiler_warnings.txt](https://github.com/jrowberg/i2cdevlib/files/3746361/MPU6050_DMP6_compiler_warnings.txt)

diyargur, 2019-10-27:
i also get the same error message!

diyargur, 2019-10-27:
Hi,
The sketch compiles cleanly in the Arduino IDE 1.8.9 using 1.6.21 board manager version.

Tanishqwayne, 2019-11-07:
hey, I have the same problem .when I compile the code, it says it's compiled but leaves so many lines of warnings and the output is affected.

jrowberg, 2021-09-26:
This is a known issue which will hopefully (finally) be fixed in the next week or so. It comes from a poor architectural decision very early in the life of the MPU6050 code. There are multiple issues addressing this, so I’m closing most of them now.

closed 2021-09-26 by jrowberg