jrowberg/i2cdevlib #486: YawPitchRoll angles reset to zero on start-up

LeonRonin, 2019-10-22:
hi I'm using your code for detecting the angle on a slope and it works just fine with very good accuracy. my only problem is that the angle values reset to zero every time restart the MCU.
is there a way to fix this??

ZHomeSlice, 2019-10-22:
I need more information to help
Do you have fixes offsets or are you using the calibration routines?
What version of the libraries are you using V2 or V6.12 of the DMP?

LeonRonin, 2019-10-22:
I used the IMU_zero sketch that comes with the library to get the offsets for my mpu6050 the I set those offsets on the V6.12 example and got very accurate readings (down to 0.01 degrees precision). my problem is that the V6.12 example resets the YAW PITCH ROLL values to zero every time the MCU starts up.

ZHomeSlice, 2019-10-24:
With fixed offsets, you can remove andy calibration routines as they will change your offsets
remark out or remove the calibration routines

```
    // Calibration Time: generate offsets and calibrate our MPU6050
    mpu.CalibrateAccel(6);
    mpu.CalibrateGyro(6);
```
These will change the fixed offsets.

trojek, 2019-11-28:
I have a similar issue. I use MPU6050_DMP6 example, I didn't use IMU_zero so I don't have fixed offsets and I don't make any changes in DMP6 example file. 
I want to get "real" values of YAW PITCH ROLL each time after restart MCU. In my case the accurance is not critical. How can I achive it?