jrowberg/i2cdevlib #491: Error compiling for board Atmel SAMD21 Xplained Pro

Dev7319, 2019-11-28:
I have tried to fix this issue by changhing the buffer lenght and I tried both 32 and 64 but it was still unable to compile.
Arduino: 1.8.10 (Windows 10), Board: "Atmel SAMD21 Xplained Pro (via EDBG)"

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp: In member function 'void MPU6050::PrintActiveOffsets()':

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: error: 'dtostrf' was not declared in this scope

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3322:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[0], 5, 0, ",  ");

  ^~~~~~~~~~~

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: note: suggested alternative: 'strstr'

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3322:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[0], 5, 0, ",  ");

  ^~~~~~~~~~~

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: error: 'dtostrf' was not declared in this scope

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3323:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[1], 5, 0, ",  ");

  ^~~~~~~~~~~

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: note: suggested alternative: 'strstr'

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3323:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[1], 5, 0, ",  ");

  ^~~~~~~~~~~

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: error: 'dtostrf' was not declared in this scope

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3324:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[2], 5, 0, ",  ");

  ^~~~~~~~~~~

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: note: suggested alternative: 'strstr'

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^
[... 67 lines cut ...]
C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3308:154: note: suggested alternative: 'strstr'

 #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt

                                                                                                                                                          ^

C:\Users\UserName\Documents\Arduino\libraries\MPU6050\MPU6050.cpp:3329:2: note: in expansion of macro 'printfloatx'

  printfloatx("", Data[2], 5, 0, "\n");

  ^~~~~~~~~~~

Multiple libraries were found for "Wire.h"
 Used: C:\Users\UserName\AppData\Local\Arduino15\packages\atmel-samd21-xpros\hardware\samd\0.3.0\libraries\Wire
 Not used: C:\Users\UserName\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire
Multiple libraries were found for "MPU6050_6Axis_MotionApps20.h"
 Used: C:\Users\UserName\Documents\Arduino\libraries\MPU6050
Multiple libraries were found for "I2Cdev.h"
 Used: C:\Users\UserName\Documents\Arduino\libraries\I2Cdev
exit status 1
Error compiling for board Atmel SAMD21 Xplained Pro (via EDBG).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Dev7319, 2019-11-28:
This is the full error i am getting. If anyone would be able to help me it would be great.

ZHomeSlice, 2019-11-28:
try adding 
#include <avr/dtostrf.h>

Dev7319, 2019-11-28:
Thanks it fixed taht error but now I am getting this error

In file included from C:\Users\Lukas\Documents\Arduino\libraries\I2Cdev/I2Cdev.h:85:0,

                 from C:\Users\Lukas\AppData\Local\Temp\arduino_modified_sketch_590553\MPU6050_DMP6.ino:49:

C:\Users\Lukas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire/Wire.h:20:0: error: unterminated #ifndef

 #ifndef TwoWire_h

 

In file included from C:\Users\Lukas\AppData\Local\Temp\arduino_modified_sketch_590553\MPU6050_DMP6.ino:59:0:

C:\Users\Lukas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire/Wire.h:20:0: error: unterminated #ifndef

 #ifndef TwoWire_h

 

Multiple libraries were found for "I2Cdev.h"
 Used: C:\Users\Lukas\Documents\Arduino\libraries\I2Cdev
Multiple libraries were found for "Wire.h"
 Used: C:\Users\Lukas\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire
Multiple libraries were found for "MPU6050_6Axis_MotionApps20.h"
 Used: C:\Users\Lukas\Documents\Arduino\libraries\MPU6050
exit status 1
Error compiling for board Atmel SAMD21 Xplained Pro (via EDBG).

Dev7319, 2019-11-28:
> 
> 
> try adding
> #include <avr/dtostrf.h>
> 
> the macro:
> `#define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {char S[(Spaces + Precision + 3)];Serial.print(F(" ")); Serial.print(dtostrf((float)Variable,Spaces,Precision ,S));}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt`
> 
> basically does a fancy fixed-length print of a floating-point number
> you could avoid using the formated float by changing the printfloatx macro
> 
> #define printfloatx(Name,Variable,Spaces,Precision,EndTxt) { Serial.print(F(Name)); {Serial.print(F(" ")); Serial.print(Variable);}Serial.print(F(EndTxt)); }//Name,Variable,Spaces,Precision,EndTxt
> 
> Or you could replace the use of the macro all together
> 
> change:
> `printfloatx("", Data[2], 5, 0, "\n");`
> 
> to
> 
> ```
> Serial.print(F(Name)); 
> Serial.print(F(" "));
> Serial.print(Data[2]);
> Serial.print(F(EndTxt));
> ```

Thanks it fixed taht error but now I am getting this error

In file included from C:\Users\UserName\Documents\Arduino\libraries\I2Cdev/I2Cdev.h:85:0,

             from C:\Users\UserName\AppData\Local\Temp\arduino_modified_sketch_590553\MPU6050_DMP6.ino:49:

C:\Users\UserName\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire/Wire.h:20:0: error: unterminated #ifndef

#ifndef TwoWire_h

In file included from C:\Users\UserName\AppData\Local\Temp\arduino_modified_sketch_590553\MPU6050_DMP6.ino:59:0:

C:\Users\UserName\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire/Wire.h:20:0: error: unterminated #ifndef

#ifndef TwoWire_h

Multiple libraries were found for "I2Cdev.h"
Used: C:\Users\UserName\Documents\Arduino\libraries\I2Cdev
Multiple libraries were found for "Wire.h"
Used: C:\Users\UserName\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\libraries\Wire
Multiple libraries were found for "MPU6050_6Axis_MotionApps20.h"
Used: C:\Users\UserName\Documents\Arduino\libraries\MPU6050
exit status 1
Error compiling for board Atmel SAMD21 Xplained Pro (via EDBG).